simprecicchiani/ThinkPad-T460s-macOS-OpenCore #3: APFS Formatted Disks

timknab, 2020-04-06:
I had a Clover setup working, which I recently removed to fully reinstall. I can boot into the MacOS installer using Opencore with the EFI provided and have used Disk Utility to format my SSD as APFS. When I get to the point in the installer where I need to choose the install volume it won't let me select the SSD and provides a message that says:

`Your Mac needs a firmware update in order to install to this volume. Please select a Ma OS Extended (Journaled) volume instead`

Have you attempted to install to an APFS formatted volume?

simprecicchiani, 2020-04-06:
That's strange, my volumes are actually formatted in APFS.

timknab, 2020-04-06:
Hmm. Did you convert a Clover EFI volume over to Opencore or fresh install?

simprecicchiani, 2020-04-06:
I converted it from Clover

simprecicchiani, 2020-04-06:
I can confirm that formatting a USB drive in APFS works on my machine.

timknab, 2020-04-06:
Very strange, I tried installing with HFS and it got further but after a few reboots said MacOS couldn't be installed to your machine due to a firmware issue or something to that effect. Oh well, guess I'll go back to Clover.

junaedahmed, 2020-04-08:
@timknab Are you using NVMe SSD or SATA SSD. I have heard that non-apple thirt party NVMe SSD's are not supported out of the box.

simprecicchiani, 2020-04-08:
@junaedahmed you’re right that could be the cause, my EFI doesn’t include any nvme fix.

timknab, 2020-04-08:
@junaedahmed  SATA SSD. I was able to install and boot using Clover, so I don't think it's that... It's a good thought, thanks!

I guess I should point out that technically it's a Thinkpad Yoga P40 with an i7-6600u and HD520, maybe there's just something really different in the firmware or bios?

timknab, 2020-04-08:
@simprecicchiani Do you know if there are any EFI nvme fixes out there? I'd be happy to add one and give it a try

simprecicchiani, 2020-04-08:
@timknab at this point it’ll be definitely something about ACPI compatibility. You have to open your original DSTD.aml and figure out if my patches make any sense for you. You can’t just suppose that it will work.

timknab, 2020-04-08:
@simprecicchiani Yeah, I've started digging through those, I have the original DSDT.aml dump from clover. I was hoping there might be an easy (or easier) solution. Thanks for your help, I'll let you know if I get it worked out.

junaedahmed, 2020-04-08:
@timknab Is you Kernel -> Quirk -> <key>ThirdPartyDrives</key><true/> in OC

timknab, 2020-04-08:
@junaedahmed Yes, I have ThirdPartyDrives set to TRUE in Kernel -> Quirk in my OC config.plist

i-arber, 2020-04-09:
> `Your Mac needs a firmware update in order to install to this volume. Please select a Ma OS Extended (Journaled) volume instead`

I've got exactly the same issue. I have a T460s. No idea how to fix this one :(

timknab, 2020-04-09:
I've been playing with all sorts of config.plist options, going through the OC laptop and desktop guides to try and figure out if it's an ACPI thing and still no luck, but I do have a Thinkpad X1 Yoga (P40) so figured it would be due to differences in ACPI or something. Weird that your'e getting it with a T460s..

timknab, 2020-04-09:
Sorry, didn't mean to close this! @i-arber, weird that you're getting the same issue with an actual T460s. I figured, for me, it was probably just differences between an X1 Yoga (P40) and the the 460s....

junaedahmed, 2020-04-09:
Hi @timknab 
I think this topic is related to your problem. Make sure your smbios bios version is latest on clover config. then you will be good to go.
`https://www.tonymacx86.com/threads/fix-for-firmware-update-required-for-mac-pro-5-1.232945/`

i-arber, 2020-04-09:
@timknab It's really, very strange. I'm using gibMacOS on Windows to make USB installer. Do you think this may be causing the issue? 

I also tried installing using HFS and after a while it did boot up the installer, but was unable to continue since it required some sort of firmware update. At that point I tried running the startosinstall command from Terminal. It started installing, restarted, showed the apple logo with loading bar below, restarted again, then got back to install screen. 

Then I tried converting the drive to APFS. Since the installer was already there, I should've been able to boot into it, but it didn't show up as a boot option in OC. So may there be some kind of error with OC loading APFS drives?

As far as my google research went, this seems to be a known issue even to real Macs. And it's usually solved by installing an older version of macOS first, then updating via App Store. Would it be a good idea to go with Sierra first, and then update? I'm unsure whether these kexts are compatible with a four-year-old version...

At this point I'm just guessing, but could it be a SMBIOS issue where OC is not reporting a MacBook Pro 13,1 but instead some older version (hence the need for "firmware update")?

simprecicchiani, 2020-04-09:
I was reached by a guy on reddit who managed to install Catalina on his T460s without problems. I should point out that he used an older EFI loaded with OC 0.5.6 and SMBIOS settings during installation. I’ve just asked him a clarification, I’ll let you know as soon as he respond.

timknab, 2020-04-10:
@i-arber I used gibMacOS on a real Mac to create the install USB so I doubt that's it. I also tried the HFS thing and the exact same thing happened to me where it seemed to be installing/working and eventually encountered a firmware error. 

It definitely seems like some weird SMBIOS firmware version issue. junaedahmed's link might have some leads, I have to do some more digging.

Thanks for looking into this a bit more, simprecicchiani

simprecicchiani, 2020-04-10:
Hey everyone, I’ve got an update.
Everything I wrote before was confirmed. Installation on APFS is possible with OC 0.5.6 and proper SMBIOS (tested on T460s)
[image]

simprecicchiani, 2020-04-12:
Quick update, check [this new build](https://github.com/simprecicchiani/Thinkpad-T460s-macOS-OpenCore/tree/master/EFI058Debug) if you want

i-arber, 2020-04-12:
@simprecicchiani thank you for the updates. I managed to bypass the installation step by using Clover. Right after the installation was finished I switched to OC and it's working flawlessly.

I did try installing using OC 0.5.6 and proper SMBIOS but it was unsuccessful. At this point, I have absolutely no idea what could be causing this issue.

I'll give your new build a try later on. Thanks a lot

hexart, 2020-04-15:
I successfully fixed this problem. Using MacBookPro13,1 machine type to generate MLB, ROM and serial number, then change machine type to MacBootPro14,2 , then try to install 10.15.4 will be quite smooth.

simprecicchiani, 2020-04-15:
I’ve just reinstalled macOS cause I made a mess with builtin packages and frameworks.
I used my main OC 0.5.7 with everything already configured (SSDTs and kexts) but SMBIOS set to MacBookPro14,2 as @hexart suggested. Installation wents perfectly fine and after adding my previous SMBIOS (MBP13,1) it’s all up and running as before.

simprecicchiani, 2020-04-16:
Another installation went fine as reported in #6 , closing the issue.

closed 2020-04-16 by simprecicchiani