simprecicchiani/ThinkPad-T460s-macOS-OpenCore #6: Installation Shut Down during Apple Logo

arthurvsmith, 2020-04-15:
Hi!

I'm having troubles installing the MacOs on my ThinkPad t460s (Intel i5 and without special WiFi card)

I first tried creating the USB by using OpenCore 0.5.7 to download MacOs 10.14.5 and then replacing the OC folder with yours including the config file. But I couldn't boot the USB.

Then I did the same, but instead of copying the config file I used the sample config file from OpenCore 0.5.7 and then manually copied all the entries from your config file. I finally was able to boot.

After the first installation round completed my laptop rebooted and after the bar under the Apple logo got to approximately 30% my laptop suddenly shut down and after that every time I try to boot I get a screen saying that MacOs could not be installed.

Am I doing something wrong? How to properly create the installation USB?

simprecicchiani, 2020-04-16:
Hi, please try the EFI folder attached below.
[EFI057Install.zip](https://github.com/simprecicchiani/Thinkpad-T460s-macOS-OpenCore/files/4485112/EFI057Install.zip)

simprecicchiani, 2020-04-16:
Be careful to select `Install MacOS` entry every time it shows up!

arthurvsmith, 2020-04-16:
> Be careful to select `Install MacOS` entry every time it shows up!

It rebooted and then I selected Installer and again after a minute or so it shut downs. I'm making a clean install and I formatted the SSD as AFTS.

arthurvsmith, 2020-04-16:
Ok, weird I rebooted a couple of times and now the bar goes further...

simprecicchiani, 2020-04-16:
You want to select the Installer the first boot only. Then you enter diskutil, format, and then start installing. After it starts, every time it reboot you have to choose "Install macOS", NOT "macOS Catalina Installer"

arthurvsmith, 2020-04-16:
It finally went through! Now, that's maybe a noob question, but do I need to do something else to the bootloader so that it stays on the laptop? It's currently using the bootloader from the USB stick I suppose.

closed 2020-04-16 by arthurvsmith

simprecicchiani, 2020-04-16:
Happy to hear that! You obviously want to move the EFI folder from USB stick to EFI partition of your main disk.

simprecicchiani, 2020-04-16:
Plus, change SMBIOS settings inside config.plist using MacBookPro13,1 model.
https://dortania.github.io/OpenCore-Desktop-Guide/post-install/iservices.html

arthurvsmith, 2020-04-16:
> Plus, change SMBIOS settings inside config.plist using MacBookPro13,1 model.
> https://dortania.github.io/OpenCore-Desktop-Guide/post-install/iservices.html

I generated a MacBookPro 14,1 serial number and Apple login works. Should I leave it or change to MacBookPro 13,1? What's the difference?