simprecicchiani/ThinkPad-T460s-macOS-OpenCore #8: How to modify CFG lock

hexart, 2020-04-28:


hexart, 2020-04-28:
[image]

EETagent, 2020-04-30:
GRUB patching is not possible on any ThinkPad with Skylake and newer. EFI is locked.
See https://github.com/taina0407/T480-OpenCore-Hackintosh/issues/4
Your only chance is to hardware flash experimental BIOS patch called Skyra1n ( Similar to famous 1vyrain )

simprecicchiani, 2020-08-15:
Hey, I've just discovered this new [guide on CFG unlocking](https://www.reddit.com/r/hackintosh/comments/hz2rtm/cfg_lockunlocking_alternative_method/?utm_source=share&utm_medium=ios_app&utm_name=iossmf). The result was the same error as with `Modified GRUB Shell` and this is `VerifyMsrE2` screen after trying to change the var.
 
Anyway I tried to boot without `AppleCpuPmCfgLock` and `AppleXcpmCfgLock` quirks and it works.
BIOS version is latest 1.49
I'm wondering if someone could test booting without `AppleCpuPmCfgLock` and `AppleXcpmCfgLock` and report if it works without touching the BIOS.

simprecicchiani, 2020-08-15:
I checked the offset with `Modified GRUB Shell` and I think it worked!
[image]
[image]

simprecicchiani, 2020-08-16:
I made a [guide](https://github.com/simprecicchiani/Thinkpad-T460s-macOS-OpenCore/blob/master/Guides/Unlock-CFG.md) for it.

closed 2020-08-16 by simprecicchiani

hexart, 2020-08-17:
I extracted the previous version 1.47 (N1CET79W) of T460s' BIOS file, the address of CFG Lock is also "0x3A", so the guide you made is also compatible with the older bios version of this type of machine.

hexart, 2020-08-17:
[image]
Any idea why ctrl+W can't save the variable?

simprecicchiani, 2020-08-17:
No idea, it works tho.

hexart, 2020-08-17:
[image]
Would you please check your CFG Lock status with Hackintool? You can check it by running Tools-AppleIntelInfo.

simprecicchiani, 2020-08-17:
It appears like your screenshot, but I am without quirks and inside RU.efi var 003A is saved as 00.

hexart, 2020-08-18:
Both Hackintool and RU.efi shows 003A is 01, so I haven't try to boot without quirks yet.

hexart, 2020-08-18:
By the way, I've tried Ctrl+W several times in RU.efi, but it went wrong saving the variable every time.

simprecicchiani, 2020-08-18:
Yeah, I have 01 on RU.efi too. I guess I was wrong and it's not possible to change it. BTW Grub Shell was wrong too at this point...
Would you try to boot without quirks? I'm without them since then and the laptop is just fine.
[image]

hexart, 2020-08-19:
For sure, I'll test it and report back.

faust86, 2020-09-01:
> 
> 
> [image]
> Any idea why ctrl+W can't save the variable? My bios version: 1.49(N1CET81W)

I have the same issue - any thoughts?

simprecicchiani, 2020-09-01:
@faust86 yep it’s not possible to change the values via software. Firmware flashing seems the only way to make it.

faust86, 2020-09-01:
So quirks is only way at this time?

simprecicchiani, 2020-09-01:
You don’t need them actually. I’m running without them for 2 weeks now and everything is fine.

benbender, 2020-11-10:
JFYI: The actual way to do it is with an SPI-flasher and these patches: http://paranoid.anal-slavery.com/biosmods/skylake.html

antoniomcr96, 2021-08-02:
Same here with Lenovo ThinkPad L390. Same error in RU.efi but system boots fine without the patches. Does anyone know what it means?

simprecicchiani, 2021-08-02:
Don't worry about it. Unlocking the BIOS has zero benefit over using patches (or not in your case).

antoniomcr96, 2021-08-14:
For who is interested, this is the “official” explaination. Unlocking CFG with haswell+ and 10.11+ shouldn’t make any difference
 
https://github.com/acidanthera/bugtracker/issues/1751

1alessandro1, 2021-08-16:
> Don't worry about it. Unlocking the BIOS has zero benefit over using patches (or not in your case).

@simprecicchiani this is wrong. Patching AppleCPUPM or AppleIntelPowerManagement is a very bad idea when CFG Lock is not possible. 

Users should check OpenCore log for MSR 0xE2 status and this helped @dreamwhite to resolve random shutdowns/panics after two hours of intense usage

For all Lenovo users, I feel you, since I owned one and they're locked up for writing very badly (modGrubshell does not work, as well as RU.efi)

The only way for lenovos to disable the bad quirks and allow native power management is to use the official InsydeH2O tool described in this guide:

https://github.com/macos86/BIOS-Manual-Mode-InsydeH2O/blob/master/GuidaInsydeH2O.md

simprecicchiani, 2021-08-16:
@1alessandro1 My comment came from personal experience, I owned a T460s for a year. Unlocking the BIOS had no effect whatsoever on user experience (same performance, temperature, issues).

I'm sure this is not the case for every system and I'm not suggesting otherwise. I just wanted to warn people trying to flash firmware mods (which is a dangerous activity and I'm no expert in it as well) since, in this specific case, it's useless.

1alessandro1, 2021-08-17:
> I just wanted to warn people trying to flash firmware mods 

In practice, using the InsydeH2O tool you're not "modding/flashing the bios, since with that tool you can modify at runtime (e.g. in windows) some settings which couldn't be accessed otherwise because of Lenovo plans of locking the end user up for no reason. I would not consider this "flashing another bios" since all you're doing is locating an offset in a certain region and changing a bit from 01 to 00. Every arch user could do this with a little bit of patience and documentation

> in this specific case, it's useless.

This needs real proof. The acidanthera team clearly explains in the manual not to use AppleXCPM patches, since they can cause instability. If this is not your case, it does not mean it's useless, since many users I helped have benefited from this unlock

VinylNerd, 2021-12-07:
@1alessandro1 im trying to figure out if i should unlock CFG for my T440P, i do need 1 quirk to boot, but i cant find one person who did it on T440P

1alessandro1, 2021-12-09:
@VinylNerd I'd extract the BIOS and find the correct offsets for your setting

https://github.com/macos86/Estrazione-BIOS-da-exe/blob/master/Estrazione_BIOS_Lenovo.md

This is the english version translated on the fly by google:
https://github-com.translate.goog/macos86/Estrazione-BIOS-da-exe/blob/master/Estrazione_BIOS_Lenovo.md?_x_tr_sl=it&_x_tr_tl=en&_x_tr_hl=en

simprecicchiani, 2021-12-10:
> @1alessandro1 im trying to figure out if i should unlock CFG for my T440P, i do need 1 quirk to boot, but i cant find one person who did it on T440P

Actually that's the model I first find a guide for. Here it is https://www.youtube.com/watch?v=ce7kqUEccUM .