simprecicchiani/ThinkPad-T460s-macOS-OpenCore #10: Right cmd key doesn't work

hexart, 2020-05-06:
After updated the newest version of EFI folder, 0.5.8 opencore, the right cmd key doesn't work at all.
And the two-finger-tap will not shows the rightmouse menu in most cases. I reversed the last dev version of VoodooPS2Controller makes it better.

simprecicchiani, 2020-05-06:
Yeah now Alt gr is mapped to Option in ACPI.
I have to agree with you it’s frustrating to cmd+del without left cmd.
I didn’t noticed trackpad issues, could you link me to the dev version you’re using?

MSzturc, 2020-05-06:
Noticed similar problems with the Touchpad on my T460. You could take the VoodooPS2 from my current build. It contains a working version which should be ~7-10 days old i guess

hexart, 2020-05-06:
[VoodooPS2Controller.kext.zip](https://github.com/simprecicchiani/Thinkpad-T460s-macOS-OpenCore/files/4586004/VoodooPS2Controller.kext.zip)
This version of VoodooPS2Controller provides better two-finger-tab response(maybe 500-700ms delay) and worse four-finger-gesture.

simprecicchiani, 2020-05-06:
> Noticed similar problems with the Touchpad on my T460. You could take the VoodooPS2 from my current build. It contains a working version which should be ~7-10 days old i guess

I'm currently using that one too, had problem while dragging windows and files with the newest one

> [VoodooPS2Controller.kext.zip](https://github.com/simprecicchiani/Thinkpad-T460s-macOS-OpenCore/files/4586004/VoodooPS2Controller.kext.zip)
> This version of VoodooPS2Controller provides better two-finger-tap response(maybe 500-700ms delay) and worse four-finger-gesture.

Did you make changes or is it just newer?

hexart, 2020-05-06:
> I'm currently using that one too, had problem while dragging windows and files with the newest one
>
>
> Did you make changes or is it just newer?

This one is a little bit newer, Compiled on one day before the official release 2.1.4

simprecicchiani, 2020-05-06:
> This one is a little bit newer, Compiled on one day before the official release 2.1.4

Ok this make sense, thanks

MSzturc, 2020-05-06:
> After updated the newest version of EFI folder, 0.5.8 opencore, the right Alt key turns into option key, it makes cmd+delete much harder than before.
> And the two-finger-tap will not shows the rightmouse menu in most cases. I reversed the last dev version of VoodooPS2Controller makes it better.

@simprecicchiani 
You should provide ACPI Sources for T460s 

@hexart 
I guess @simprecicchiani is using my ACPI Keyboard which patches right alt since it's needed for my keyboard which has a german layout. You could simply  comment out

`"e038=e05b", //AltGr=Left Windows`

in [here](https://github.com/MSzturc/Lenovo-T460-OpenCore/blob/master/ACPI/SSDT-KBRD.dsl) and rebuild the SSDT ( or use the attached one here )

[SSDT-KBRD.aml.zip](https://github.com/simprecicchiani/Thinkpad-T460s-macOS-OpenCore/files/4586358/SSDT-KBRD.aml.zip)

hexart, 2020-05-06:
> @simprecicchiani
> You should provide ACPI Sources for T460s
>
> @hexart
> I guess @simprecicchiani is using my ACPI Keyboard which patches right alt since it's needed for my keyboard which has a german [...]

It works to comment //"e038=e05b" in SSDT-KBRD.aml file.
Anyway, thanks so much to both of you! @simprecicchiani @MSzturc

closed 2020-05-06 by hexart

simprecicchiani, 2020-05-10:
Pushing keyboard customisation to the next level ahahah
[image]

hexart, 2020-05-11:
> BTW @MSzturc thanks for F4 LED indicator!

Have you tried [Thinkpad Assistant](https://github.com/MSzturc/ThinkpadAssistant/releases)? It's a brilliant idea. If it's possible, you can try to provide Italian language pack.

simprecicchiani, 2020-05-11:
> Have you tried [Thinkpad Assistant](https://github.com/MSzturc/ThinkpadAssistant/releases)? It's a brilliant idea. If it's possible, you can try to provide Italian language pack.

Of course I am! Where can I contribute to language translation?

simprecicchiani, 2020-06-02:
Newer version of VoodooPS2 are still  unreliable. I did't had time to go through plist settings, did you guys managed to fix it?

hexart, 2020-06-02:
Haven't tried yet. I will keep you posted.

hexart, 2020-06-02:
I tested touchpad with the newest version of VoodooPS2Controller official release. It's the same as before. Two-finger tap response well.
But, as always, as soon as you use little red dot and the 3 mouse button on top of the touchpad even once, the multi-finger gesture goes no response any more unless you restart the computer.

hexart, 2020-06-03:
> Also, click-and-drag with touchpad only works fine but with physical button the movement and animation lags tremendously.

Does it lags in Windows?