simprecicchiani/ThinkPad-T460s-macOS-OpenCore #11: Sleep wake issue

ariffathur, 2020-05-20:
Hello, first I would to thanks for your work on this project, I am newbie on hackintosh

Yesterday I was trying to install catalina using your efi057install on my thinkpad t460, everything goes fine, but I have a sleep wake issue.

My laptop goes to sleep for about a second and goes to wake again suddenly, even worst sometimes its goes to black sceen if I am in hidpi mode using RDM. Is anyone face the same issue? or know what should do might be hepfull for me, thanks

simprecicchiani, 2020-05-20:
Hi! You’ll probably need to [map your USB ports](https://github.com/simprecicchiani/Thinkpad-T460s-macOS-OpenCore/blob/master/README.md#usb-ports-map).

MSzturc, 2020-05-20:
You could follow the guide or use my EFI it has an USBMap for the Laptop Ports and the Docking Station

https://github.com/MSzturc/Lenovo-T460-OpenCore

simprecicchiani, 2020-05-20:
> You could follow the guide or use my EFI it has an USBMap for the Laptop Ports and the Docking Station
> 
> https://github.com/MSzturc/Lenovo-T460-OpenCore

That would be even better

ariffathur, 2020-05-20:
okay, thanks for the suggestion, I will try

closed 2020-05-20 by ariffathur