simprecicchiani/ThinkPad-T460s-macOS-OpenCore #13: Batteries can't be recognized under windows booting by open core

hexart, 2020-06-06:
[image]
[image]

simprecicchiani, 2020-06-06:
That was predictable, I didn’t bother with  Windows-sensible patches since using it wasn’t in my plans.  
If you want to implement solutions for it feel to fork the repo and pull request 👍🏻

simprecicchiani, 2020-06-06:
Moreover, current SSDT and renames patches are from @MSzturc T460 repo, which I think are based on @junaedahmed's T460 repo.

hexart, 2020-07-20:
https://github.com/daliansky/OC-little/blob/master/08-电池补丁/08-1-Thinkpad/各机型电池补丁/SSDT-NTFY.md

This might be the solution of two physical batteries machine type, FYI.

junaedahmed, 2020-07-20:
@hexart 
use this one [SSDT-BATC](https://github.com/acidanthera/VirtualSMC/blob/master/Docs/SSDT-BATC.dsl) for dual battery in windows.

closed 2020-08-05 by hexart