simprecicchiani/ThinkPad-T460s-macOS-OpenCore #14: HDMI no signal while mini DP only work sometimes

winteraspect, 2020-06-15:
Ciao Caro, the repo works fine except for external monitor. Neither HDMI nor mini DP works fine. HDMI just cannot output any signal while DP can only hold for seconds. Maybe it's the Catalina's fault?

aidbal, 2020-06-23:
Hey @simprecicchiani , first of all, thanks for the repo! I have succesfully installed Catalina on my T460s with I5-6300. Everything works fine.

However, I have the same issue as @winteraspect -  the HDMI output does not work. I am using HDMI -> DisplayPort to connecto to my monitor.

After connecting the cable, the mouse flashes some times, everything kind-of slows down (looks like it's preparing to output video to hdmi), but after that, nothing happens - black screen. Monitor reports that there is no signal coming.

I can try to fix it myself and create a pull request. Can you guide us where we should start looking for the issues?

simprecicchiani, 2020-06-23:
Hi guys, sorry for being late in answering. First of all I'm well aware of the issue but I rarely use external monitors, when I do I use either a miniDP to HDMI or HDMI to HDMI.
They both don't work when plugged in: the external monitor shows the picture while the internal goes black. This could be fixed by going on preference panel > monitor settings > push and hold ALT > click on "find monitor". By doing this the internal monitor starts displaying content too. Closing the lid and reopen it might help too.
The issue is, as far as my knowledge goes, a matter of WhateverGreen framebuffer patches. 
I followed (this guide)[https://www.tonymacx86.com/threads/guide-general-framebuffer-patching-guide-hdmi-black-screen-problem.269149/] to get HDMI audio working (with the tweaks descibed above) in the first place.
It should be a trial and error process to get them working properly (trying different connector IDs, find the right flags, etc.).
If I'm not mistaken on the T460s the config is as follows:
| con | port |
| --- | --- |
| 0 | internalDP |
| 1 | HDMI |
| 2 | miniDP |

winteraspect, 2020-06-24:
Thanks for the reply. The "find monitor" solution may not apply since I have sometimes both screen go black. My workaround is to avoid the situation that both screen working. So when I use external screen, I close the lid. The tricky thing is that I often need to open the lid for 2 seconds to have the external monitor light up. I only use displayport now so not sure if the workaround applys to HDMI. For HDMI, the external monitor never light up while the "displaypolicyd" process consumes high CPU usage. No results from Google. Looking forward to any updates on these minor issues. The repo still works like a charm and better suit my needs than Clover. Grazie mille!

aidbal, 2020-06-24:
I have tried following WhateverGreen FAQ, especially this section: [LSPCON driver support to enable DisplayPort to HDMI 2.0 output on IGPU]( https://github.com/acidanthera/WhateverGreen/blob/master/Manual/FAQ.IntelHD.en.md#user-content-lspcon-driver-support-to-enable-displayport-to-hdmi-20-output-on-igpu). But could not get HDMI output to work.

Just to clarify some things - my external monitor does not light up at any point. My internal screen is always lit up, regardless if HDMI is connected or not.

simprecicchiani, 2020-06-24:
@aidbal Could you share with us your current framebuffer patches?

aidbal, 2020-06-24:
Sure.

I have tried 0, 1 and 2, as connector ids. 
I have also tried playing with adding boot arg `-igfxlspcon` or adding `enable-lspcon-support` to IGPU.

[image]

simprecicchiani, 2020-06-24:
I would try a fresh start without `conX-alldata` and then incrementally adding properties for con2

aidbal, 2020-06-24:
Alright, I finally got it working.

From IGPU dict, I removed:
`framebuffer-con2-enable`
`framebuffer-con2-alldata`
I added:
`framebuffer-con2-has-lspcon` with Data `01000000`
`enable-lspcon-support` with Data `01000000`

The only caveat is you cannot boot with hdmi cable plugged in. If you do so, both internal and external displays will go black. If you boot and then connect HDMI cable, everything works as expected.

Either way, thank you for the help!

[image]

simprecicchiani, 2020-06-24:
How about audio passthrough? is it working?
At that point you might as well remove the other connectors data as I'm pretty sure they are injected by WG defaults with `16190000` id.

aidbal, 2020-06-24:
No, unfortunately, no output devices (except for Built-in) are showing up in  Settings -> Sound -> Output.

Can confirm that HDMI output is still working after removing other connectors' data:

[image]

simprecicchiani, 2020-06-27:
Hi guys! I've just updated the repo with my latest config, HDMI is working perfectly with audio!

[image]

[image]

MiniDP to HDMI is working too but without audio, I think it should have the same config as `con2` in order to function.

winteraspect, 2020-07-04:
Good news is that I have added some boot-args from whatevergreen and now it's more stable. "-igfxhdmidivs -igfxlspcon". Please read the manual for more info, https://github.com/acidanthera/WhateverGreen.

simprecicchiani, 2020-07-14:
> Good news is that I have added some boot-args from whatevergreen and now it's more stable. "-igfxhdmidivs -igfxlspcon". Please read the manual for more info, https://github.com/acidanthera/WhateverGreen.

I tried them but they actually made it worse. The internalDP don't resume displaying when disconnected from external monitors.

cty12, 2020-11-17:
@simprecicchiani 

Hi Simone,

First thank you for the hard work for getting the configuration working! I wonder if the 

> Internal monitor turns black when external is connected

known issue that is listed in `README.md` related to this problem? On my machine (T460s with FHD) the **HDMI** hotplug works fine (using your `0.6.3` configuration) however the **Mini DP** doesn't - plugging it in causes the internal display to go black and the internal screen will not turn on even if the external (Mini DP) monitor is disconnected. 

This is somewhat unfortunate since otherwise the Thinkpad docking station works even without any USB injector kext. Since the display signal on the docking station is DP, it suffers from the same issue with the Mini DP port as is described above.

simprecicchiani, 2020-11-19:
Hi, could you please try to change framebuffer-con1-type to 00080000?
You may also need to change the port count and add a connector, but try this first.

cty12, 2020-11-19:
@simprecicchiani 

> Hi, could you please try to change framebuffer-con1-type to 00080000?

I guess you mean `00040000`, which is for **DP** right? If I get it right, in your current configuration both frame buffers (`con1` and `con2`) are set to be `00080000` which stands for **HDMI**. I've tried setting:

1. `con1` to be DP 
2. `con1` and `con2` both be DP

However in either cases, the connection is very unstable both under HDMI and DP so I reverted back to your configuration, which at least makes HDMI functional. 

Interestingly, on my Macbook Pro late 2013, both frame buffers are set to be DP, even if I connect to an HDMI display.

cty12, 2020-11-20:
@simprecicchiani 

I did a few more experiments with an external display:

1. Boot the machine with internal display off and HDMI cable plugged in. This causes the internal display to remain black after HDMI disconnects and the only workaround to turn on the internal display again (without rebooting) is to open the lid, unplug HDMI, close the lid and then re-open the lid. 

2. It does not work at all if the displayport (I use the one on the docking station but I guess the MiniDP on the laptop itself would be the same) connects to a passive DP to HDMI converter (using dual-mode DP) and then to the monitor.

simprecicchiani, 2020-11-20:
As I said, the behavior changes with OS’s and driver updates. miniDP to HDMI connection has worked since day one; with the latest updates, as you reported, it doesn’t anymore!

cty12, 2020-11-20:
> As I said, the behavior changes with OS’s and driver updates. miniDP to HDMI connection has worked since day one; with the latest updates, as you reported, it doesn’t anymore!

Oh because I did a fresh install with Big Sur so I didn't realize that it once worked! Thank you for reporting that.

ozanhonamlioglu, 2020-12-13:
> Alright, I finally got it working.
>
> From IGPU dict, I removed:
> `framebuffer-con2-enable`
> `framebuffer-con2-alldata`
> I added:
> `framebuffer-con2-has-lspcon` with Data `01000000` [...]

Hi, may be this can help the issue that "cannot boot with hdmi cable plugged in"

I found this in the document of WhateverGreen
"you might need temporarily to enable the infinite loop fix before connecting a HDMI monitor to your build, otherwise the system just hangs, so you won't be able to run the IORegistryExplorer and find the connector index."

https://github.com/acidanthera/WhateverGreen/blob/master/Manual/FAQ.IntelHD.en.md#fix-the-infinite-loop-on-establishing-intel-hdmi-connections-with-a-higher-pixel-clock-rate-on-skylake-kaby-lake-and-coffee-lake-platforms

wrobeljakub, 2021-01-29:
> > As I said, the behavior changes with OS’s and driver updates. miniDP to HDMI connection has worked since day one; with the latest updates, as you reported, it doesn’t anymore!
> 
> Oh since I did a fresh install with Big Sur, I didn't realize that it once worked! Thank you for reporting that. 
So, which macos version is verified to work with HDMI monitors?

mhl221135, 2021-01-29:
> So, which macos version is verified to work with HDMI monitors?

i am using bigsur with hdmi monitor and it is working fine sound over hdmi also working

wrobeljakub, 2021-01-29:
> i am using bigsur with hdmi monitor and it is working fine sound over hdmi also working

Could you try turning display off and on with HDMI cable plugged in? In my case the display locks up and I have to replug the HDMI cable.

mhl221135, 2021-01-29:
> > > > > As I said, the behavior changes with OS’s and driver updates. miniDP to HDMI connection has worked since day one; with the latest updates, as you reported, it doesn’t anymore!
> > > > 
> > > > 
> > > > Oh since I did a fresh install with Big Sur, I didn't realize that it once worked! Thank you for reporting that.
> > > 
> > > So, which macos version is verified to work with HDMI monitors?
> > 
> > i am using bigsur with hdmi monitor and it is working fine sound over hdmi also working
> 
> Could you try turning the display off and on with HDMI cable plugged in? In my case the display locks up and I have to replug the HDMI cable.

Everything ok vithout repluging (maybe different monitors i have dell one)

mhl221135, 2021-08-21:
I dont shure but maybe solution can be related wit EDID profile https://github.com/tylernguyen/x1c6-hackintosh/issues/60

lanhuai, 2021-09-12:
I found a way to get an 4k external display to work properly by using Display Port.
By following these steps:

1. get the laptop to boot up
2. Login to the desktop
3. Put the laptop into hibernation
4. Plug the Display Port cable of the external monitor into the laptop
5. Wake up the laptop

Now, the external monitor lights up normally.

> My laptop is ThinkPad X260.
> I have done the BIOS Mod, 'DVMT-Prealloc' has been set to 64MB.

This is my screenshot:

[image]

onestrongside, 2021-11-11:
So I installed macOS Monterey and everything works flawlessly right out of the box with the latest release (0.7.3) except for the miniDP. I have been fiddling around with this for a couple of days and I just can't get my 4K display (LG 27UL500-W) to show up with the correct resolution of 3840x2160 and shows up as 2560x1440. I have it connected from miniDP to HDMI. I tried everything from SwitchResxX to messing around with EDID to changing the framebuffer in the config.plist but to no avail. I hope I don't have to mod my BIOS because I can't get my hands on the necessary tools...

mintyleaf, 2022-11-19:
> * get the laptop to boot up
> * Login to the desktop
> * Put the laptop into hibernation
> * Plug the Display Port cable of the external monitor into the laptop
> * Wake up the laptop

for my HDMI dock port just putting laptop to sleep and waking up after reboot just works fine.
even if I unplug it from the dock after this - HDMI still working

but, if I unplug laptop from the dock, put to sleep, wake, and THEN put to dock - it stops working, so need to resleep it again while sitting in the dock

it will be nice to find exact point at the sleep/wake process where display port fixes and implement that as workaround script
but I'm pretty new in this hackintosh topic, not sure if I can figure it out by myself =D

I remember that I had kinda similar (just proves that DP is a problematic) issues with DP in linux - it continue to identify as new DP with new ID every replug from dock, so it was hard to automate window setup by the script