simprecicchiani/ThinkPad-T460s-macOS-OpenCore #15: Low Trackpad sensitivity

aidbal, 2020-06-24:
Hey,

I have noticed that trackpad sensitivity is kind of low. What I mean is that sometimes it registers three finger swipe, sometimes it does not. I have to kind of press more firmly compared to a real mac.

I am not talking about "Tracking speed" which just makes your mouse move faster.

Is there anyway I could tweak this? I tried scavenging through various forums, but people only talk there about "Tracking speed" not the actual sensitivty of registering your fingers.

simprecicchiani, 2020-06-24:
Yeah three finger swipe works only 50% of the times if you are not used to the kind of sensibility it has.
You should look into VoodooPS2 repo to get familiar with how it works, I have never tried to change anything.

aidbal, 2020-06-25:
Alright, so after fiddling a bit, I have found this package: https://github.com/acidanthera/VoodooPS2. Using it now for a day, it seems like it made the trackpad more responsive (especially the 3 fingers swipe).

Closing it now :)

closed 2020-06-25 by aidbal

simprecicchiani, 2020-06-25:
What do you mean by this package? Acidanthera’s VoodooPS2 kext is already built into my repo’s EFI.
You updated it?

aidbal, 2020-06-25:
Oh, okay, I thought you were using Rehab's (https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller). 

To answer your question, yeah, I downloaded the latest version (2.1.5)

simprecicchiani, 2020-06-25:
Newer version are known to cause some issues and lag when performing some actions as discussed in #10 (second part). At least for me it makes all choppy when dragging windows around using physical keys and Trackpoint. But yeah, swipes gestures are smoother in the latest version. Didn’t you noticed some weird behaviours with it?

simprecicchiani, 2020-06-25:
In addition 2.1.5 will break PrtScr key map if you have one, unless you don’t modify the kext itself.

simprecicchiani, 2020-08-06:
Hey, I've just updated the repo with a new trackpad kext, It's a lot snappier than before and supports all gestures!