simprecicchiani/ThinkPad-T460s-macOS-OpenCore #21: About changing 4k  screen

xumanwei, 2020-07-23:
Thank you for your work! Everything worked fine before I changed my screen from 1080p to 4k.  The problem is that the display is ok in win10 but can not be lighted  in macos. I have changed the framebuffer value but it can not work.  I supposed there are some kernel crashes due to the value of dvmt preallote  but I don't know how to solve this question, I am so grateful if you can help me with this question.

hexart, 2020-08-05:
hd520 doesn't support 4k officially, I did also change to 4k, then gave up and changed back to 2k.

hexart, 2020-08-26:
Try to set platform-id: 193b0005 of your hd520/530 for 4k screen.

xumanwei, 2020-08-26:
> Try to set platform-id: 193b0005 of your hd520/530 for 4k screen.

I tried just now, the screen can be lighted but the core graphics  is not driven. The system shows the memory of graphic is only 31Mb

MSzturc, 2020-08-26:
4k wont work when internal display is activated. For me it only worked via displayport (it might be not working since hotplug isnt stable). Try to set Boot display to dp in bios and Boot with External Monitor

xumanwei, 2020-08-27:
> 4k wont work when internal display is activated. For me it only worked via displayport (it might be not working since hotplug isnt stable). Try to set Boot display to dp in bios and Boot with External Monitor

Thank you, I will try it later.

rwecho, 2020-09-23:
@xumanwei  how's going on now? 
Hope you can resolve it.

hexart, 2020-09-23:
The only way to make hd520 support 4k screen is changing dvmt to 64M in bios, it can solve the issue once and for all.

xumanwei, 2020-09-23:
> The only way to make hd520 support 4k screen is changing dvmt to 64M in bios, it can solve the issue once and for all.

According to the information of the internet, the original dvmt value in bios of  thinkpad t460s may be 128M?

xumanwei, 2020-09-23:
> > According to the information of the internet, the original dvmt value in bios of thinkpad t460s may be 128M?
> 
> I modded the bios and default value is 32MB.
> I’m currently running 64MB without patches but haven’t had the occasion to try 4K.

Wonderful! Could you tell me how to set the dvmt?, I had a try , but the bios seems to be locked

simprecicchiani, 2020-10-09:
> Wonderful! Could you tell me how to set the dvmt?, I had a try , but the bios seems to be locked

[BIOS Mod Guide](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/blob/master/Guides/Bios-Mod.md)

xumanwei, 2020-10-09:
> > Wonderful! Could you tell me how to set the dvmt?, I had a try , but the bios seems to be locked
> 
> [BIOS Mod Guide](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/blob/master/Guides/Bios-Mod.md)

So niceeee!!! thinks a lot,I will try it later ,but maybe some wrong occurs in the website: 
     http://paranoid.anal-slavery.com/files/backup_thinkpads/xx60_patches_v1.txt

simprecicchiani, 2020-10-15:
Here you go [xx60_patches_v1.txt](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5383993/xx60_patches_v1.txt).

xumanwei, 2020-10-16:
> Here you go [xx60_patches_v1.txt](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5383993/xx60_patches_v1.txt).

Thanks! After I moded bios’ Pre-allocated to 128M, then I added some fb to the config.plist.Then 4k screen works. 

I added the following things :
[image]

closed 2020-10-16 by xumanwei

simprecicchiani, 2020-10-16:
Great, would you mind sharing the picture again?

usman2006, 2021-05-01:
> > Here you go [xx60_patches_v1.txt](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5383993/xx60_patches_v1.txt).
> 
> Thanks! After I moded bios’ Pre-allocated to 128M, then I added some fb to the config.plist.Then 4k screen works.
> 
> I added the following things :
> ![Uploading 4Q~C71{N`EUNSHK_TV3{55.png…

Hi, 
i´m also trying to get a 4k monitor to work.
can you show me your config that i need for 4k.

thx

xumanwei, 2021-05-02:
> Hi,
> i´m also trying to get a 4k monitor to work.
> can you show me your config that i need for 4k.
>
> thx

Ok, I will try to find the original  file, but I am not sure whether it can work ,this is because I cleaned my disk one time which my lead to the loss of the file. this is the file I  have find, wish this file can help you. [Click here](https://drive.google.com/file/d/1TTld0QeyPLwUtjELCteym2b0yM5Wu1rC/view?usp=sharing), if it cannot work, tell me and I will try to find another version.

usman2006, 2021-05-14:
> Ok, I will try to find the original file, but I am not sure whether it can work ,this is because I cleaned my disk one time which my lead to the loss of the file. this is the file I have find, wish [...]

sry for the late  reply.
i´ve got kernal panic and than reboot.
dont know if this i likend to that but i use a dockingstation and right now i have two external display. one 4k over DP and  the secound  over vga.
somehow the both  showing only as one...mean always in sync mod. and locked in 1080p

dseknat50, 2025-01-23:
I have followed the steps mentioned in the doc to change mod the bios and I am able to see the bios advanced options. Could someone tell me how to set dvmd to 64 MB.
If you someone can tell the navigation path in bios to set this value. It would be helpful.