simprecicchiani/ThinkPad-T460s-macOS-OpenCore #26: Bluetooth and WLAN Adopter does not work...

sanghyukYoon, 2020-09-08:
I was using Bluetooth and wlan adapter well in the clover environment() but after switching to this accessore, I used the existing keystrokes but they didn't work at all.
Also, I don't know why batteries don't change at 1%.
Thank you

[image]
[image]
[image]

simprecicchiani, 2020-09-08:
Did you performed a full wipe of clover stuff? Delete all third party kexts and rebuild cache?

simprecicchiani, 2020-09-08:
Also, I don't know which GUI software you're using to open the configuration but it looks like you have some Voodoo kexts enabled when they shouldn't.

duszmox, 2020-09-08:
It's called OpenCore configurator, and it's a ton of bullcrap. It messes with your config.plist if you don't have the correct version which matches with your OC version, and I also read that even if you have the correct one it could cause serious damage to the config file. You should really use propertree or some other plist editor.

duszmox, 2020-09-08:
Also if the battery status doesn't work, you should replace the SMCBatteryManager with the RehabMan's ACPIBatteryManager.

sanghyukYoon, 2020-09-09:
Yes I wiped all

closed 2020-10-09 by simprecicchiani