simprecicchiani/ThinkPad-T460s-macOS-OpenCore #27: Monitor resolution after Stop

alexdelli, 2020-09-10:
Hi, when system restart after Stop the image on the screen in half than normal. So I have to restart
I have 1920x1080 resolution and HDPI enabled
thank you

simprecicchiani, 2020-09-11:
I don't use this method to get HiDPI. Last time I checked it worked fine on my wqhd display.
Try to revert the changes and use this method:
- Disable SIP (just for this process, you can enable it once finished)
- Download and install [RDM Utility](https://github.com/usr-sse2/RDM/releases)
- Launch the app, click on "resolution", then "edit" (or use listed resolution if preferred)
- Type 2880x1620, check HiDPI (will look like 1440x810) (or use whatever resolution is compatible)
- Reboot the system
- Re-launch RDM, click on "resolution", select 1444x810⚡️

[image]

alexdelli, 2020-09-11:
Ah ok, before I used the RDM utility without issue.
So I can close this issue.
Just a question, since the last update when I start the "macbook" appear the logon screen, and then the verbose mode. Is it correct ?
Good job, I wish to update to big sure asap

simprecicchiani, 2020-09-11:
You can change your boot arguments as you like. I ‘ll always leave oc picker and verbose ON inside the distributed configuration.
Once you’re up and running it’s safe to turn everything “debug related” off.

alexdelli, 2020-09-11:
ok, thank you again

closed 2020-09-11 by alexdelli