simprecicchiani/ThinkPad-T460s-macOS-OpenCore #28: Sreen Flicker on lower brightness

SamuelG13, 2020-09-15:
Hi. I successfully instaled macOS Catalina on my ThinkPad, everything works great except the screen. When the brightness is at maximum there is no flicker but at lower brightness, the flicker appears and I can't use the laptop it just hurts my eyes. Thank you in advance !

simprecicchiani, 2020-09-15:
Can you post your specs?

SamuelG13, 2020-09-17:
Non-touchscreen 1920x1080p 60hz screen, core i5-6200u, 16gb ram, sandisk ssd

simprecicchiani, 2020-09-19:
Sorry, can't help you with that. Try to ask on r/Hackintosh or somewhere else.

simprecicchiani, 2020-10-30:
Please try [this release](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/releases/tag/0.6.2.1).

simprecicchiani, 2020-11-05:
I assume it has been fixed like #30 .

closed 2020-11-05 by simprecicchiani