simprecicchiani/ThinkPad-T460s-macOS-OpenCore #29: Wifi for Intel chip

alexdelli, 2020-09-17:
Hi, sorry this is not an issue, but a request of implement a guide for intel wifi. Now with AirportItlwm.kext is it not necessary to use Heliport. Maps, Handoff , Airport are working (Big Sur and Catalina).  It's very simple, just put AirportItlwm.kext in EFI and ADD Kernel in config.
What do you think about it?

simprecicchiani, 2020-09-17:
I didn't see that coming! It seems great news, I'll check it out as soon as I have some free time.

alexdelli, 2020-09-17:
> Edit: I'm thinking of buying a T490 

Noooooooooooo, we need you againnnn :((

simprecicchiani, 2020-09-19:
Kexts are in (disabled by default, make sure to enable them).
Waiting for your feedback.

alexdelli, 2020-09-19:
For me it's ok! Fantastic

sanghyukYoon, 2020-09-21:
I did it and it works well

sanghyukYoon, 2020-09-21:
Here is the link https://github.com/1hbb/OpenIntelWireless-Factory

simprecicchiani, 2020-09-21:
Could you make a list of what works and what don't?

- [ ] Handoff
- [ ] Universal Clipboard
- [ ] iPhone Cellular Calls
- [ ] Text Message Forwarding
- [ ] Instant Hotspot
- [ ] Auto Unlock
- [ ] Continuity Camera
- [ ] Continuity Sketch
- [ ] Continuity Markup
- [ ] Sidecar
- [ ] AirDrop
- [ ] Apple Pay

 As of https://support.apple.com/en-us/HT204681

alexdelli, 2020-09-22:
- [ ] Handoff
- [ ]   Universal Clipboard
- [ ]   iPhone Cellular Calls
- [ ]   Text Message Forwarding
- [ ]   Instant Hotspot
- [ ]   Auto Unlock
- [ ]   Continuity Camera
- [ ]   Continuity Sketch
- [ ]   Continuity Markup
- [ ]   Sidecar
- [ ]   AirDrop
- [ ]   Apple Pay

> Could you make a list of what works and what don't?
>
> * [ ]  Handoff
> * [ ]  Universal Clipboard
> * [ ]  iPhone Cellular Calls
> * [ ]  Text Message Forwarding
> * [ ]  Instant Hotspot
> * [ ]  Auto Unlock
> * [...]

simprecicchiani, 2020-10-09:
Waiting for new updates from you guys with Intel WLAN

closed 2020-10-09 by simprecicchiani