simprecicchiani/ThinkPad-T460s-macOS-OpenCore #30: Screen brightness levels t460s fullHD touch

mhl221135, 2020-09-30:
The brightness levels switch incorrectly and the first half of the slider makes the screen completely black during the first sleep cycle of the laptop, it returns to normal and then breaks again = (
Also, the touch screen does not work, but it's not scary (=

my spec:
t460s
1920x1080 touch
12gb ram
i5 6300u
intel WLAN
intel 240GB SSD

mhl221135, 2020-09-30:
Also, after a long sleep, the laptop freezes when trying to turn on and only reboot helps = (

mhl221135, 2020-09-30:
Also maybe you know what is it ?) Why it not disabled ?[image]

simprecicchiani, 2020-10-09:
You may have to adjust the patches. I've never owned a touchscreen display so I can't help you.

mhl221135, 2020-10-09:
> You may have to adjust the patches. I've never owned a touchscreen display so I can't help you.

you think that brightness trouble depends on that my screen is touch?(

simprecicchiani, 2020-10-09:
> you think that brightness trouble depends on that my screen is touch?(

I don't really know, the sleep freeze problem might be.

simprecicchiani, 2020-10-20:
Please try this [EFI](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5409170/EFI.zip).

mhl221135, 2020-10-21:
> Please try this [EFI](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5415066/EFI.zip).
> 
> Edit: attached EFI changed

My screen is black with this EFI but system is working(i hear sounds of Catalina when pressing keyboard)

simprecicchiani, 2020-10-30:
Please try [this release](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/releases/tag/0.6.2.1).

mhl221135, 2020-11-01:
> Please try [this release](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/releases/tag/0.6.2.1).

This pre-release has fixed the brightness slider
but kext stopped working for intel wi-fi (I can not test the laptop normally because I work with it)

how can I bring the brightness fix to your stable release (or just wait for the next one)?

I still don't know if you've seen, but I found a solution for an external monitor (add a key and plist igfxonln = 1) so the external monitor works after sleep, but after reboot, only reconnection or a sleep cycle helps

simprecicchiani, 2020-11-01:
> but kext stopped working for intel wi-fi (I can not test the laptop normally because I work with it)

No idea why that happened. Everything but batteries and display stuff should be unchanged.

> how can I bring the brightness fix to your stable release (or just wait for the next one)?

Just swap `SSDT-PNFL.aml` and add `AppleBacklightSmoother.kext`

> I still don't know if you've seen, but I found a solution for an external monitor (add a key and plist i`gfxonln=1`) so the external monitor works after sleep, but after reboot, only reconnection or a sleep cycle helps

I saw that but I haven't had the occasion to test it. Would you mind sharing the source of this fix? I looked at WEG manual but haven't found anything extensively mentioning it.

mhl221135, 2020-11-01:
https://www.reddit.com/r/hackintosh/comments/fw16mt/fix_for_10154_hdmi_wake_from_sleep/

Whatevergreen release 1.3.8 

`Added igfxonln=1 boot argument and force-online IGPU property force online status for all displays`

simprecicchiani, 2020-11-01:
> https://www.reddit.com/r/hackintosh/comments/fw16mt/fix_for_10154_hdmi_wake_from_sleep/
> 
> Whatevergreen release 1.3.8
> 
> `Added igfxonln=1 boot argument and force-online IGPU property force online status for all displays`

Yeah I saw that, I expected a little bit more documentation for it. Anyway, if it does work it's great!

closed 2020-11-01 by simprecicchiani

mhl221135, 2020-11-01:
Also u tell about brightnes, also how i can test new battaries patch (maybe you can tell me which files to replayce) thank you.