simprecicchiani/ThinkPad-T460s-macOS-OpenCore #31: Shutdown when the first battery is discharged

mhl221135, 2020-10-09:
460s
i5 5300u
12 GB RAM
Intel SSD 240
WLAN / BT Intel

The laptop does not use both batteries at the same time (

Shutdown when the first battery is discharged
And then, when I turn it on, it shows me that the battery is 100% discharged and works until the second battery is discharged (

how can i fix it? on 6.0.0 this was not (
all i did on update:
1) deleted the EFI folder from the partition 2) threw your latest release instead
3) plist replaced platform info with your own
4) and included intel bluetooth and wifi in plist
5) nvme power managment also is enabled

(all thi things is from your default confid.plist my changes only: platform-info, intelWlan and BT, and NVME power)

simprecicchiani, 2020-10-09:
Try this guide https://github.com/daliansky/OC-little/tree/master/08-电池补丁/08-1-Thinkpad

simprecicchiani, 2020-10-15:
You could also try older version of VirtualSMC and related plugins

simprecicchiani, 2020-10-20:
Take a look at #13 too.

closed 2020-10-20 by simprecicchiani

simprecicchiani, 2020-10-20:
Please try this [EFI](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5409170/EFI.zip).