simprecicchiani/ThinkPad-T460s-macOS-OpenCore #32: Logged out when running resource extensive applications

meshack007, 2020-10-14:
Hello,
Thank you for the detailed instructions, i was able to install MacOS Catalina on ThinkPad T460s

I have been using it for a while but everytime i open up a resource extensive applications or processes the PC begins to overheat and logs itself out

These include building apps, opening multiple emulators and the like.
Is there any solution to this ?
Thanks.

faust86, 2020-10-14:
+1 for me

simprecicchiani, 2020-10-15:
In `config.plist` under `Kernel->Quirks` try setting `AppleXcpmCfgLock` to `True`.
I also changed the method inside SSDT-PLUG in the latest commits according to Acidanthera guidelines. I wonder if it could cause some problem on certain machines. This is the old one [SSDT-PLUG](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/blob/c8a323e9b71a3090e7d52986ccdfe1b5d6d5bf46/EFI/OC/ACPI/SSDT-PLUG.aml).

meshack007, 2020-10-18:
Thanks, the config.plist update helped with the overheating. Now when idle i think i have more free CPU than before. But it still logs out when i open heavy apps.

I also changed the SSDT-PLUG but it did not help. I think i had the older one too.

simprecicchiani, 2020-10-19:
What do you mean by logs out? Are you using custom power management or voltage shifting?

meshack007, 2020-10-20:
By logs out i mean: The screen becomes black for at-least a second, after that the login screen is displayed. when i log in again everything that was opened before is closed and the os starts re opening them.

I installed NVMe and CPUFriendFriend.command from the installation instructions, nothing else for power management.

simprecicchiani, 2020-10-20:
Try to remove CPUFriend and companion data. Use plain SSDT-PLUG and report back please.

hirwa145, 2020-10-20:
I am having this issue also

simprecicchiani, 2020-10-20:
You can try an older version like [0.6.0](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/releases/tag/0.6.0).

simprecicchiani, 2020-10-30:
Did anyone tested if a rollback help with this issue?

closed 2020-10-30 by simprecicchiani

meshack007, 2020-10-31:
> Did anyone tested if a rollback help with this issue?

Yes, i tested with the 0.6.0 and 0.6.2 release but the problem still persist. The laptop gets hot and logs itself out.

KenWuqianghao, 2020-11-14:
I was able to solve this issue simply by adding a 16GB ram stick to my laptop. Not sure why.