simprecicchiani/ThinkPad-T460s-macOS-OpenCore #34: Battery readouts not working

adorableteemo, 2020-10-16:
Battery readouts seems to not work OC 0.6.2 using ur EFI files. It's stuck on 1% and the status says battery is not charging.

zalevsk1y, 2020-10-20:
Have the same problem. 
__________________________
lenovo t460, i5 6200u. Os 10.15.7

simprecicchiani, 2020-10-20:
Please try this [EFI](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5409170/EFI.zip).

adorableteemo, 2020-10-21:
> Please try this [EFI](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5409170/EFI.zip).

This EFI fixes it, thanks a lot! Keep up the amazing work!

simprecicchiani, 2020-10-21:
It would be useful if you can share your DSDT so I can see which patches you needed.

faust86, 2020-10-28:
I stick ith my question here :) Does Mac OS shows 2 battery? it is almost certain that the system only uses one battery - the time is much lower than in Windows.

mhl221135, 2020-10-28:
> I stick ith my question here :) Does Mac OS shows 2 battery? it is almost certain that the system only uses one battery - the time is much lower than in Windows.

This config is patched to use both batteries but its combine both and system see its like one battery

simprecicchiani, 2020-10-30:
I will release the EFI with all the patches but it would be better if someone could provide a DSDT to work on.

closed 2020-10-30 by simprecicchiani

zalevsk1y, 2020-11-01:
> Please try this [EFI](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5409170/EFI.zip).

Thank you. This EFI works very good.

mhl221135, 2020-11-04:
> I will release the EFI with all the patches but it would be better if someone could provide a DSDT to work on.

if you still need it, this is my files extracted by different ways(you can choose which you like)
[extracted_tables.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5486037/extracted_tables.zip)