simprecicchiani/ThinkPad-T460s-macOS-OpenCore #37: Use dual monitor, must unplug/replug HDMI after sleep

ibnuzamra, 2020-10-24:
Just setup a new hackintosh that is running great! Except for this one thing

When I came to my T460s this morning, it has initiated Display Sleep but I couldn't wake it up, even though the laptop was running fine otherwise.

I had to unplug/replug the HDMI cable from my laptop and then it woke up. That seems pretty specific – does anyone know how to fix this?

mhl221135, 2020-10-24:
I have the same problem

P.s. this problem listed in readme.md in "not working section"

simprecicchiani, 2020-10-24:
I tried everything as far as my knowledge goes but I couldn't fix it.

mhl221135, 2020-10-26:
I was helped by adding the boot key igfxonln = 1, the monitor also does not turn on after booting the system, but it wakes up after sleep

soeryobadja, 2020-11-02:
@ibnuzamra Seems like you have external display working, based on the issues you submitted? My T460s has the same symptom described in this guide, i.e. internal display turns black when connecting to external display, nothing shown in the external display, and internal display keeps black when external display disconnected. Any modification you do to the files? Thanks.

Great files by @simprecicchiani that enables a very stable T460s used with Catalina 10.15.7.

simprecicchiani, 2020-11-05:
@soeryobadja it's fixed since release 0.6.2.2 thanks to @mhl221135 

Please use the [latest release](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/releases/latest).

closed 2020-11-05 by simprecicchiani