simprecicchiani/ThinkPad-T460s-macOS-OpenCore #38: Brightness Control issu, CPUFriend can't find Data, some errors at booting

usman2006, 2020-10-24:
hi, 
first big thx for providing yours files.
i have t460s 6300u, 12Gb, 512GB, FHD with Touch, Dockingstation.

I used the EFI fron issu "Battery readouts not working" and mist of things are working fine.
So the first problem I notice is that my screen get completely at dark (off) at 75% level through key or the slider in the settings.
The second problem is that want to use CPUFriend to get the low freq. (800mhz I mostly use on battery) as without I get only 1300mhz. I followed your instructions and also search on other form but it don't work.
I always get an critical halt error. OC: Image Kexts\CPUF.. is missing for injected kext  CPUFriend.kext (Frq.. data injector).
i´v tried with dataprovider kext and also apci  but always get that error.
and the last problem is that I get some error at booting but its boot normal.

thx for your help in advance.

simprecicchiani, 2020-10-30:
Did you turn CPUFriend.kext ON on your config.plist? It's set to `Enabled=false` by default.

usman2006, 2020-10-30:
yes i did

simprecicchiani, 2020-10-30:
My bad, I just found an issue inside `config.plist`

[image]

This line is wrong. The executable doesn't have an extension. It should be `Contents\MacOS\CPUFriend`

closed 2020-10-30 by simprecicchiani