simprecicchiani/ThinkPad-T460s-macOS-OpenCore #40: Did you use SSD or PCIe NVME, i want to upgrade to PCie NVME that works with Catalina/BigSur. Can you please specify which Storage did you use if it NVMe?

hirwa145, 2020-10-27:


simprecicchiani, 2020-10-28:
My T460s is equipped with a SATA SSD but you'll be fine upgrading to NVMe. Just don't use the few models that are known to be faulty with macOS (like Samsung PM981).

hirwa145, 2020-10-29:
Which model do you suggest i can use? For 512Gb Storage

hirwa145, 2020-10-29:
After looking on internet, i found that SAMSUNG 970 EVO 500G NVME PCIe M.2-2280 SSD is compatible with the laptop, but want to make sure if it wont cause any problems with hackintosh

hexart, 2020-10-29:
I‘m using westen data nvme ssd, its perfect, but a little bit high power consumption. 
[image]

simprecicchiani, 2020-10-30:
> After looking on internet, i found that SAMSUNG 970 EVO 500G NVME PCIe M.2-2280 SSD is compatible with the laptop, but want to make sure if it wont cause any problems with hackintosh

As mentioned in the [anti-mackintosh buyers guide](https://dortania.github.io/Anti-Hackintosh-Buyers-Guide/Storage.html), unless the proper firmware is installed, a 970 EVO won't run on macOS.

closed 2020-10-30 by simprecicchiani