simprecicchiani/ThinkPad-T460s-macOS-OpenCore #42: pre-release 6.2.1 and 6.2.2 intel wifi issues

mhl221135, 2020-11-02:
In this pre-release, Intel Wi-Fi stops working (as in 6.2.1)

I transferred `AppleBacklightSmoother.kext` and `SSDT-PNLF.aml` to the 6.2 stable release there the brightness works fine as well as intel wi-fi kext

here is the log from release 6.2.2

`02:211 01:914 OC: Prelinked injection AirportItlwm.kext (Intel WiFi driver) - Invalid Parameter`
[opencore-2020-11-02-090621.txt](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5473934/opencore-2020-11-02-090621.txt)

simprecicchiani, 2020-11-02:
I'm pretty sure it isn't a configuration issue, rather a kext injection problem (cache or permission on your specific machine).

mhl221135, 2020-11-02:
> I'm pretty sure it isn't a configuration issue, rather a kext injection problem (cache or permission on your specific machine).

Maybe you know how can i try to fix it ?

simprecicchiani, 2020-11-03:
I'm not sure right now. In the meantime you can easily replace all `ACPI -> Patch` from pre-release EFI to your `config.plist` and replace `SSDT-BATT`, `SSDT-BATC` and `SSDT-NTFY` with `SSDT-BAT0`, `SSDT-BATC` and `SSDT-NTFY` from pre-release as well (changing the names accordingly in `ACPI -> Add`.

mhl221135, 2020-11-03:
I ran these commands but wi-fi also does not work (

sudo chmod -Rf 755 / S * / L * / E *
sudo chmod -Rf 755 / L * / E *
sudo chown -Rf 0: 0 / S * / L * / E *
sudo chown -Rf 0: 0 / L * / E *

sudo kextcache -i /

ill try your last recomendation later and tell you, wery big thank you for your job

mhl221135, 2020-11-04:
> I'm pretty sure it isn't a configuration issue, rather a kext injection problem (cache or permission on your specific machine).

I find solution with wifi on latest pre-release

[https://github.com/daliansky/XiaoMi-Pro-Hackintosh/wiki/Drive-Native-Intel-Wireless-Card#use-airportitlwm]

I changed `SecureBootModel` to `Default` and `DmgLoading` already were changed to `Signed` now my wifi properly working

there are also other solution methods there, but I have not tried them since this one helped me

closed 2020-11-04 by simprecicchiani