simprecicchiani/ThinkPad-T460s-macOS-OpenCore #43: The battery displays the wrong parameters.

xuandung38, 2020-11-02:
The battery displays the wrong parameters. Not with the remaining capacity. therefore there will be a situation where there is no low battery notification resulting in a sudden shutdown.

xuandung38, 2020-11-02:
My origin DSDT + SSDT Here. 
Bios was mod

[origin.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5475497/origin.zip)

simprecicchiani, 2020-11-02:
Which release are you on?

xuandung38, 2020-11-03:
latest version bro!

simprecicchiani, 2020-11-03:
Try [this](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/releases/tag/0.6.2.2) pre-release.

xuandung38, 2020-11-03:
Still this error! :(

mhl221135, 2020-11-04:
Hi, the last pre-release still has battery problems
the laptop is working fine and HWmonitorSMC shows the real capacity of my battery (I have very tired batteries, but there are about 2500mAh) during the discharge, the capacity continues to fall and near about 40% of the charge knocks out the message that the laptop will soon fall asleep and at this point the percentage stop changing, and the HWmonitorSMC shows 2500mAh and after a while the laptop just turns off )=

it seems that at the moment of notification of a low capacity, the laptop ceases to see the real battery capacity

xuandung38, 2020-11-30:
Sloved

closed 2020-11-30 by xuandung38