simprecicchiani/ThinkPad-T460s-macOS-OpenCore #44: Random Reboots when the fan starts working

KenWuqianghao, 2020-11-04:
First-time Hackintosh user and I followed the guide successfully till the end. But when I use Hackintosh do some tasks that are a little bit heavier, the hackintosh reboots.

simprecicchiani, 2020-11-05:
Please try the [latest release](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/releases/tag/0.6.3).

KenWuqianghao, 2020-11-05:
I installed my hackintosh yesterday, so I'm one hundred percent certain it is the newest version.

simprecicchiani, 2020-11-05:
I released an updated version few hours ago. Anyway, it would be useful to see some logs when the reboot occurs.

KenWuqianghao, 2020-11-05:
So generally I faced three situations that forced reboots, one is the attached image, saying memory status killing idle process. The second one says something about applekeymanagement. I wasn't able to get a image of the second one and the third simply doesn't display anything and reboots. 
[image]

simprecicchiani, 2020-11-05:
Don’t you have some system logs when the system reboots? Which are your specs?

KenWuqianghao, 2020-11-05:
My specs are 512gb ssd, 4gb ram, intel GPU, and intel core i7. For the system logs, I'm not sure which part I should get, give me a moment to read through them.

mhl221135, 2020-11-06:
Hello again))
I have some issues when running photoshop or other resource addicted programs my laptop sometimes go to log out and reopen programs whe i login in

i have aportunity to make this discusting photography but my knowlage in mac os is still miserible, so maybe you find the way to fix this issue

here is my system log which i copy from hackintool 
[err.txt](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5502435/err.txt)

and here is discusting screenshot with info apearing when system loggin out)
[image]

KenWuqianghao, 2020-11-06:
> Hello again))
> I have some issues when running photoshop or other resource addicted programs my laptop sometimes go to log out and reopen programs whe i login in
>
> i have aportunity to make this [...]

Same, I faced this issue severely too, especially I only have 4gb of Ram.

meshack007, 2020-11-13:
I had the same issue #32 but after i added RAM it has helped. although it might not be a solution your looking for.

mhl221135, 2020-11-14:
> I had the same issue #32 but after i added RAM it has helped. although it might not be a solution your looking for.

I alredy have 12gb of memory (also now i am on big sur and dont see this issue this time)

meshack007, 2020-11-14:
@KenWuqianghao can you check your console and see if on crash reports you have WindowServer crashing? Maybe ram is not the problem

KenWuqianghao, 2020-11-14:
> @KenWuqianghao can you check your console and see if on crash reports you have WindowServer crashing? Maybe ram is not the problem

I added 16gb of ram and the issue is solved.

closed 2020-11-18 by KenWuqianghao