simprecicchiani/ThinkPad-T460s-macOS-OpenCore #45: 6.2.2 pre-release wrong battery percentage when discharging

mhl221135, 2020-11-04:
Hi, the last pre-release still has battery problems
the laptop is working fine and HWmonitorSMC shows the real capacity of my battery (I have very tired batteries, but there are about 2500mAh) during the discharge, the capacity continues to fall and near about 40% of the charge knocks out the message that the laptop will soon fall asleep and at this point the percentage stop changing, and the HWmonitorSMC shows 2500mAh and after a while the laptop just turns off )=

it seems that at the moment of notification of a low capacity, the laptop ceases to see the real battery capacity

_Originally posted by @mhl221135 in https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/issues/43#issuecomment-721755279_

simprecicchiani, 2020-11-05:
I don't know what to say anymore. I even looked into your DSDT and it is basically the same as mine. I've never experienced such problems. I'm wondering if it has something to do with the BIOS. Which version are you on?

mhl221135, 2020-11-05:
i have last version of bios from lenovo site(not patched)

xuandung38, 2020-11-05:
like me. before have this error #43

mhl221135, 2020-11-30:
Its not ocuring anymore, if its ocure again i reopen this issue, thanks for all

mhl221135, 2020-12-24:
On latest release this issue ocures again(

simprecicchiani, 2020-12-26:
Battery patches were untouched. Try rolling back to a previous release of VirtualSMC (and related plugins).

mhl221135, 2020-12-29:
> Battery patches were untouched. Try rolling back to a previous release of VirtualSMC (and related plugins).

It is not helps (

mhl221135, 2021-01-07:
On latest comments this issue still exist (

I also noticed that when the charge is connected, the percentage sometimes jumps, for example, 100 - I connect the charger and it starts to show 60 as well hwmonitorsmc shows different maximum capacity after reboot

simprecicchiani, 2021-01-07:
Do you remember which release was correctly working for you?

mhl221135, 2021-01-07:
> Do you remember which release was correctly working for you?

я думаю, что это 0.6.3 но сейчас не уверен)

simprecicchiani, 2021-01-07:
> i think it's 0.6.3 but i'm not sure now)

It would be a helpful info if you can test it.

mhl221135, 2021-01-07:
> > i think it's 0.6.3 but i'm not sure now)
> 
> It would be a helpful info if you can test it.

I will test and tell you

mhl221135, 2021-01-07:
> > i think it's 0.6.3 but i'm not sure now)
> 
> It would be a helpful info if you can test it.

I just checked, on 0.6.3 there is still this problem ((I probably mistakenly decided that it disappeared since I worked at the office most of the time from charging

soeryobadja, 2021-01-08:
> Do you remember which release was correctly working for you?

Jumping in to share my experience with battery indicator with older release of 0.6.0.

I have my battery percentage relatively working well (I might say so) while still using 0.6.0 version. 
Here is my experience.
1. Battery indicator come up to 100% when charged completely. 
2. Discharging from 100% to 99% is done at a relatively slow rate, comparing with measurement from HWMonitor the battery already discharging. The battery indicator only start discharging to 99% when the indicator at HWMonitor already around 95% and then both keeping the pace of discharging since then. 
3. When the battery is discharging and nearly empty, it does not show low battery alert which makes the laptop suddenly goes off without any warning.    
4. When I plug the charger in low battery condition, under 5%, for a certain amount of time the indicator shows that the battery is not charging and the indicator jumps to at around 20+%. Then, after few minutes, the indicator return to normal showing the battery is charging and the battery indicator comes back into more proper percentage. 
5. When charging back up, the difference in percentage between HWMonitor and battery indicator is still at around 4-5%, eventually catching up into 100% on both. I noticed one incident where the battery charging rate was very slow but that comes after multiple cycle of computer sleep and restarting the laptop restore the charging functionality. 

I did not do any upgrade yet on the new releases, as I once tried 0.6.3 at that time but it makes my laptop booting considerably slow (issue #49 ) therefore I revert back to 0.6.0 where I noticed the boot time is at best. I am currently still at Catalina 10.15.7 19H2 and not yet upgraded and my BIOS version is 1.3.4 (I plan to upgrade the BIOS to make sure it has the same version to see if it affects this issue). 

Thanks.

mhl221135, 2021-01-08:
> Jumping in to share my experience with battery indicator with older release of 0.6.0.
>
> I have my battery percentage relatively working well (I might say so) while still using 0.6.0 version.
> Here is [...]

I am not shure that 0.6.0 support big sur and i have no ability to install catalina now to test (

simprecicchiani, 2021-01-08:
I have looked once more into the [dsdt you shared](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/issues/34#issuecomment-721554327) and it is indeed the same as mine on BIOS 1.49. 
I'm really sorry this is not working for you guys but I don't know where to look anymore.

mhl221135, 2021-01-08:
> I have looked once more into the [dsdt you shared](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/issues/34#issuecomment-721554327) and it is indeed the same as mine on BIOS 1.49. 
> I'm really sorry this is not working for you guys but I don't know where to look anymore.

Can you tell which battaries yo have (model numbers)

simprecicchiani, 2021-01-08:
> Can you tell which batteries yo have (model numbers) this is my battaries info from AIDA64

I don't have windows installed, do you know if AIDA64 runs on linux?

> maybe its nvram issue ? Looks like its appear after sleep cycle

It could be.

> also i see that bios 1.50 available should i update?

I don't think it will change anything but eventually I'll update too.

mhl221135, 2021-01-08:
> I don't have windows installed, do you know if AIDA64 runs on linux?
>
>
> It could be.
>
>
> I don't think it will change anything but eventually I'll update too.

I use boot usb with windows maybe its other programs for lunux like aida64 (i just think maybe its different hardware issue)

mhl221135, 2021-01-09:
also hwmonitorsmc and Ioreg (in hex) sometimes show differnt charge cycles and max capacity, sometimes it 2130 sometimes it 2600 etc. What can it be?
[image]

mhl221135, 2021-01-09:
May it be the problem in `SSDT-NTFY.aml`? I have not yet figured out how to edit aml and dsl, but I looked at the version from release 0.6.0 where it seems there was no problem with the battery and the latest version

In version 0.6.0, BAT0 is mentioned `External (_SB_.PCI0.LPC_.EC__.BAT0.B0ST, IntObj)`

and in the version of the latest commits it is not mentioned at all

simprecicchiani, 2021-01-11:
@mhl221135 You can find the new config.plist in the repo. Please try and report back.

mhl221135, 2021-01-13:
> @mhl221135 You can find the new config.plist in the repo as well as the SSDTs. Please try and report back.

I tried new patches, i using HwMonitorSmc2 for info:

1)Cycles count now showing correct seem

2)Battery percent still differ from mac os panel but a little less then on previous patch

3)ioregexplorer now showing both batteries names and models splited by "/"

4)Laptop still turn off when first battery discharged
(maybe problem in NTFY?)
I test it about few days with old patch and new one now i am shure that laptop turn off when first battery discharged, looks like it not handle second battery when first discharged or its dont handle low percent of first battery
Maybe soluthion is to change something in code minimum percantage of first BAT for switching to second BAT?

5)Laptop still issue that fall asleep at low percantage but its still just turn off(

6) I see that you update SSDT-DYTC.dsl but its only in repo/files and its in dsl format and as i understand its related with yogasmc "quiet/balance/performance" swithcer (it is still not switching from Balance position)?

Anyway great thank you for your job, it is my first mac os expirience and you do it easier and better and better day by day))

mhl221135, 2021-01-19:
I think this is my hardware problem so i closing this issue, i already purchased new battery 00hw023 and when i recieve it i tell here the results, thank you for your great work

closed 2021-01-19 by mhl221135

mhl221135, 2021-01-20:
Yes new 00hw022 battery resolve my issue (this is the battery that drains first on our laptops)