simprecicchiani/ThinkPad-T460s-macOS-OpenCore #48: Slow Wi-Fi speed using Intel cards

D35P4C1T0, 2020-11-05:
I noticed that with my Intel 8260NGW card, whilst connected to a 5Ghz wifi, the max speed I was able to achieve were 50-60 MBit download and not much more than 20 Mbit upload. 
This same card can handle 500+ MBit, as I've tested with a different OS on the same machine.  
This might be caused by the immature kexts for Intel wifi, as my internet connection peaks out at 990 MBit-ish and my current wireless router can deliver up to 600 Mbit.

simprecicchiani, 2020-11-05:
Last time I checked the current intel WiFi driver for macos support 2.4ghz band only.

D35P4C1T0, 2020-11-05:
I'm glad 5Ghz band is somehow supported now. I'll check if transfer speeds with 2.4 GHz are more in line with the speeds I normally get.

simprecicchiani, 2020-11-05:
Original apple wlan card 100% worth IMO, works just like a real mac. I have no experience with intel wifi.

D35P4C1T0, 2020-11-05:
I'm a bit concerned about linux support tho. Might just give it a try and see.

soeryobadja, 2020-11-05:
I am using BCM94360NG which works natively without any kext with M.2 form factor so no need for adapter like Apple original card. It works in 2.4 and 5GHz in my case.

mhl221135, 2020-11-28:
Hello, i find the way to little speed up boot process for those who use intel wifi
https://github.com/OpenIntelWireless/itlwm/issues/353

If your card is AC8260 you need to replace 

`find itlwm/firmware/ -type f ! -name 'iwm-7265-*' -delete`

by

`find itlwm/firmware/ -type f ! -name 'iwm-8000C*' -delete`

Here is ready to use kexts for BigSur  
[itlwm and airportitlwm kexts.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5610989/itlwm.and.airportitlwm.kexts.zip)

P.S. Use only one kext in the same time( i recomend to use `Airportitlwm.kext`)

mhl221135, 2021-01-13:
> I noticed that with my Intel 8260NGW card, whilst connected to a 5Ghz wifi, the max speed I was able to achieve were 50-60 MBit download and not much more than 20 Mbit upload.
> This same card can [...]

try a new mentod desctibed in "Readme.md "method 2

simprecicchiani, 2021-01-13:
> try a new mentod desctibed in "Readme.md "method 2

I simplified the documentation a little, it is now under the same Intel WLAN section. You can add a brief explanation to highlight the differences between the two drivers if you'd like to.

mhl221135, 2021-01-13:
> > try a new mentod desctibed in "Readme.md "method 2
> 
> I simplified the documentation a little, it is now under the same Intel WLAN section. You can add a brief explanation to highlight the differences between the two drivers if you'd like to.
> 

As i understand difference only in driver source openbsd and intel

simprecicchiani, 2021-01-14:
> As i understand difference only in driver firmware source openbsd and intel

I meant performance differences

mhl221135, 2021-01-14:
I have been using itlwm.kext / airportitlwm.kext for several days now
and this is what I can say:
1) The speed seems to be the same as on itlwm.kext / airportitlwm.kext I can’t check because my provider only gives 100 megabits of Internet and the download speed from the router is 70 and the upload speed is 40

2) There are no reconnections as it was on itlwm.kext / airportitlwm.kext

3) After sleep, too, there are no more problems with loading the driver and connecting to the network

4) The system boot speed seems to be the same, but the verbal readings of the system contain less information about the driver than itlwm.kext / airportitlwm.kext
My system info:
Intel® Dual Band Wireless-AC 8260
Big Sur 11.1
t460s i5 6300u/ 12gb / 500gb WD SN750

mhl221135, 2021-04-10:
Now using latest intel kext and maximum speed that i can reach is 80mbit upload and download but i have no opportunity to test faster wifi spot(also replacing of you wlan card by compatible one can resolve your trouble or you need to wait for IntelOpenWireless improvment of their project)

> I noticed that with my Intel 8260NGW card, whilst connected to a 5Ghz wifi, the max speed I was able to achieve were 50-60 MBit download and not much more than 20 Mbit upload.
> This same card can [...]

closed 2021-04-10 by mhl221135