simprecicchiani/ThinkPad-T460s-macOS-OpenCore #49: Slow boot time when using 0.6.3 version

soeryobadja, 2020-11-06:
I tested using 0.6.3 version, but got very slow boot time, i.e. 1min15sec when using 0.6.3 compare to 20sec when using 0.6.0.
My device is T460s i5-6300u 8GB SSD SATA 256GB with BIOS version 1.3.4 dated 2018-03-22.

Another issue I faced is regarding battery management. Correctly showing and managing the battery consumption, but when it reaches 5% it shows a low battery warning then all of a sudden the system shutdown. Putting the charger afterwards, the system shows a 30% battery capacity, lighting indicator, but saying "Battery is not charging". A couple of restart fix this though. 

Thanks.

mhl221135, 2020-11-06:
I have the same time on 6.2 and 6.2.2 pre relase but now i upgrade efi to 6.3 an its feeling like laptop booting faster 
P.S. I think this long booting time ocured by intel wifi kext
Also sometimes have issues with battary but most time its working correct

soeryobadja, 2020-11-06:
I don't use Intel card, so no kext regarding Intel Wifi in my case. My kext folder is more compact, as I remove some of the kext I do not use, compounded by using native WiFi card that requires no WiFi+BT kext at all.
Also, do you have the card reader working? My laptop uses RT522, which is not guaranteed to work based on the readme on Sinetek-rtsx.

mhl221135, 2020-11-06:
i did not try card reader still i have no camera or other devices with SD cards

> I don't use Intel card, so no kext regarding Intel Wifi in my case. My kext folder is more compact, as I remove some of the kext I do not use, compounded by using native WiFi card that requires no [...]

soeryobadja, 2020-11-07:
Tried again today to drain the battery, and it works correctly. No sudden shutdown when battery level touching 5%. I see that the display at HWMonitor now is closer to the Apple battery indicator. So I have no more battery issues using 0.6.0 version. 

Have not tried the 0.6.3 version again for the slow boot time.

taktiker12, 2020-11-20:
You may be asking two different things, but in the last paragraph you said:

The ultimate goal is to prevent the system from automatically logging into a network
Prevent System From Automatically Logging Into a Network:

I think the way to accomplish that is to un-check (disable) the Auto-Join feature for each WiFi network name/SSID that's stored. System Preferences, Network, Advanced.... Uncheck all checked boxes in the "Preferred Networks" list. enter image description here

Also note that there are a couple of other options here that may be of interest:

You can uncheck the option to "Remember networks this computer has joined"
You can check the box to "Require administrator authorization to: Turn Wi-Fi on or off.
To Disconnect WiFi Power at Boot Time:

To disconnect wifi power at boot time, you can use cron. It's certainly simpler than building a .plist, and it's invoked at boot instead of login. In my testing, it survives a login, but of course it could be "overruled" by other system settings -- YMMV. It can be re-started from the WiFi icon in the menu bar after login when you wish. Here's how to do this from the CLI:

crontab -e 
This will start your designated crontab editor, and open your current crontab. If you don't have a crontab (haven't used it previously), you'll be prompted to select from one of three choices for default editor. I'd suggest you choose pico or nano. That done, the default crontab will be opened in the editor of your choice.

Once the editor has started, make the following entry on the first empty line at the "bottom" the file:

@reboot /usr/sbin/networksetup -setairportpower en0 off
Save and exit the editor, and your new crontab becomes active. This command will now execute each time your system boots, and (subject to the exception mentioned) your WiFi interface will be "OFF" until you turn it on.

closed 2021-05-26 by soeryobadja