simprecicchiani/ThinkPad-T460s-macOS-OpenCore #50: 6.2-6.3 EFIs issue when running photoshop or other resource addicted programs

mhl221135, 2020-11-06:
Hello again))
I have some issues when running photoshop or other resource addicted programs my laptop sometimes go to log out and reopen programs whe i login in

i have aportunity to make this discusting photography but my knowlage in mac os is still miserible, so maybe you find the way to fix this issue

here is my system log which i copy from hackintool 
[err.txt](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5502435/err.txt)

and here is discusting screenshot with info apearing when system loggin out)
[image]

_Originally posted by @mhl221135 in https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/issues/44#issuecomment-723184966_

mhl221135, 2020-11-06:
I can try to repeat the situation and make more logs, but just tell me how to write down the logs?

KenWuqianghao, 2020-11-06:
> I can try to repeat the situation and make more logs, but just tell me how to write down the logs?
Not sure how you can do that, first time too.

meshack007, 2020-11-13:
How much RAM do you have installed in your laptop ? @mhl221135

KenWuqianghao, 2020-11-13:
I had this issue when I had 4GB ram, but when I slap in my 16GB ram card the issue disappeared but replaced with a green screen when the ram card moves in the backboard.

mhl221135, 2020-11-13:
> How much RAM do you have installed in your laptop ? @mhl221135 

I have 12gb

meshack007, 2020-11-13:
> > How much RAM do you have installed in your laptop ? @mhl221135
> 
> I have 12gb

I had 8gb before when i upgraded to 12gb the issue stopped, but in this case i am not sure

meshack007, 2020-11-13:
> I had this issue when I had 4GB ram, but when I slap in my 16GB ram card the issue disappeared but replaced with a green screen when the ram card moves in the backboard.

12GB solved it for me

meshack007, 2020-11-14:
I upgraded too, but it seems it just stalls the problem because for whatever reason i can not update to BigSur, my laptop keeps logging out, there is plenty of RAM and CPU left and its not even overheating but it logs out evertime, maybe its the HDD

meshack007, 2020-11-14:
@simprecicchiani @KenWuqianghao @mhl221135 so i have been trying to look for a solution online and i have find out out the same issue occurs to due to a process known as "windowserver" crashing due to some problems in GPU, i opened up console and when i viewed my crash reports the process has several crashes as expected but i still have no solution

mhl221135, 2020-11-14:
> @simprecicchiani @KenWuqianghao @mhl221135 so i have been trying to look for a solution online and i have find out out the same issue occurs to due to a process known as "windowserver" crashing due [...]

Try to download and install from usb device

meshack007, 2020-11-14:
Okay, but wouldnt that delete everything else that i have ?

mhl221135, 2020-11-14:
> Okay, but wouldnt that delete everything else that i have ?

If yourbcatalina working now you can try i also have troubles when i try tobupdate from working system so when i use usb installer its update successfully on last efi but install/uodate process takes near 2 hours

mhl221135, 2020-11-14:
> Okay, but wouldnt that delete everything else that i have ?

If your catalina working now you can try i also have troubles when i try tobupdate from working system so when i use usb installer its update successfully on last efi but install/uodate process takes near 2 hours

simprecicchiani, 2020-11-14:
> Okay, but wouldnt that delete everything else that i have ?

Yes, it will.

Could you post your device properties for iGPU?

meshack007, 2020-11-14:
@simprecicchiani okay, i will soon enough, i have installed bigsur but my Wifi card is not being detected, is this a problem is bigsur or its just me?

simprecicchiani, 2020-11-15:
I’m sorry, I meant confit.plist’s device property.
Why do you have a 768p display? Didn’t the T460s ship with 900p, 1080p and 1440p only?

meshack007, 2020-11-15:
> I’m sorry, I meant confit.plist’s device property.
> Why do you have a 768p display? Didn’t the T460s ship with 900p, 1080p and 1440p only?

No problem, Its right here 
[Config.txt](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5542663/Config.txt)

Yeah, i kinda wrecked the display it shipped with, so in the meantime i am dealing with that. Could that be a problem ?

simprecicchiani, 2020-11-15:
> No problem, Its right here
> [Config.txt](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5542663/Config.txt)
> 
> Yeah, i kinda wrecked the display it shipped with, so in the meantime i am dealing with that. Could that be a problem ?

No it's not an issues, I was just wondering.

Can you please try this one: [Config.txt](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5543047/Config.txt) (It has all of your personal settings unchanged, just change the extension back to .plist)

meshack007, 2020-11-15:
Thank You. Done.

If i am not wrong you have removed updates with the HDMI Monitor not waking up after sleep, However i still had this issue since before that. After installing big sur and using your last release it seems pretty stable. Thanks.

simprecicchiani, 2020-11-16:
Yeah external monitors are a nightmare since 10.15.4 you can try to add back `force-online:01000000`. I made a couple of changes in there, let me know how it performs.

meshack007, 2020-11-19:
@mhl221135 do you still get the issue?

mhl221135, 2020-11-19:
> @mhl221135 do you still get the issue?

On big sur its not happend but i even dont know maybe it is Catalina trouble)

meshack007, 2020-11-19:
Same here, and i really tried! I guess this can be closed then!

meshack007, 2020-11-19:
> Can you please try this one: [Config.txt](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5543047/Config.txt) (It has all of your personal settings unchanged, just change the extension back to .plist)

This update caused my PC to shutdown every time i connected to HDMI, but the issue doesn't occur anymore with Big Sur so i have reverted back to original Config.plist

simprecicchiani, 2020-11-19:
Yeah I removed some patches form it to remove any possible conflict. But yeah, after your experiences we can say that the issues definitely had something to do with a specific Catalina release.
Off Topic: I don’t know about you guys but to me Big Sur feels just a tiny bit slower sometimes, it lack some snappiness from a mature OS. Also, some changes they made to the UI constantly drive me crazy.

closed 2020-11-19 by simprecicchiani

meshack007, 2020-11-19:
> Yeah I removed some patches form it to remove any possible conflict. But yeah, after your experiences we can say that the issues definitely had something to do with a specific Catalina release.
> Off [...]

Haha maybe this is me coming from the buggy catalina but i actually think that this one is a bit faster, even on boot times!

meshack007, 2020-11-19:
Should we have like a discord ? telegram channel ? signal ? @mhl221135 @KenWuqianghao @simprecicchiani @meshack007

simprecicchiani, 2020-11-20:
Boot time is indeed faster. Sorry but I have very little spare time. Moreover, I’m really thinking about replacing the t460s now that I’m finding myself using it tons of hours. The screen flickering to sustain lower brightness is hurting my eyes so bad.