simprecicchiani/ThinkPad-T460s-macOS-OpenCore #51: CPUFriend issue

hirwa145, 2020-11-07:
I want to do a custom CPU Power Managment, 
After using this "**git clone https://github.com/fewtarius/CPUFriendFriend; cd CPUFriendFriend; chmod +x ./CPUFriendFriend.command; ./CPUFriendFriend.command**" and entering preferred values. **ssdt_data.aml** is not provided under results folder. Only **Mac-473D31EABEB93F9B.plist** is provided? Is there something i am doing wrong?

simprecicchiani, 2020-11-07:
I think this happen when the script is unable to download IASL. You can download 
[iasl.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5505415/iasl.zip), extract it and place it in `~CPUFriendFriend/Scripts`. Then run `CPUFriendFriend.command` and it should compile the ssdt.

hirwa145, 2020-11-08:
I tried again with the solution you suggested, but i cant still find **ssdt_data.aml** file