simprecicchiani/ThinkPad-T460s-macOS-OpenCore #53: AirportItlwm on BigSur

mhl221135, 2020-11-09:
Hello, this is not an issue but i just want to inform that AirportItlwm

not working corectly with Securebootmodel=Disabled at BigSur
But itlwm + Heliport working fine (exept slow speed on both methods)

SpacialLoser, 2020-11-11:
Yeah i had slow speeds on 2.4ghz try switching to 5ghz for better speeds?

mhl221135, 2020-11-14:
> Yeah i had slow speeds on 2.4ghz try switching to 5ghz for better speeds?

No i checked speed not change when i use 5ghz

Also latest appleitlwm alpha 1.2 working fine (still slow) but works

simprecicchiani, 2020-11-15:
Big Sur can boot with SecureBoot = Default since OC 0.6.3.
There are two version of AirportItlwm included in the release, make sure you are using the big-sur one.

closed 2020-11-15 by simprecicchiani

mhl221135, 2020-11-15:
I use Latest alpha AirportItlwm with securebootmodel=disabled and it is working fine

simprecicchiani, 2020-11-15:
It's recommended to set `SecureBoot` at least to `Default` as of OpenCore documentation to run Big Sur.