simprecicchiani/ThinkPad-T460s-macOS-OpenCore #54: Unable to spoof the MAC address

KenWuqianghao, 2020-11-11:
I tried several methods including terminal and external apps, but I wasn't able to spoof my MAC address. Any solutions?

KenWuqianghao, 2020-11-16:
@simprecicchiani Hello, can you please test if Bootcamp windows can spoof mac address? I've been having a lot of trouble trying and It is a must-have feature for me. Please help me. I have done it in Hackintosh, so it should work in Bootcamp windows too.

simprecicchiani, 2020-11-17:
I'm not sure of what you're talking about. Do you need your ethernet or wlan MAC address?

KenWuqianghao, 2020-11-17:
Basically, I installed Hackintosh and used Open core to multiboot since I only have one SSD. I was able to spoof my mac address on Hackintosh, but the mac address won't spoof on bootcamp windows The address just stays the same.

simprecicchiani, 2020-11-19:
Sorry but that’s out of my knowledge, I hope someone with proper understanding will help you.

KenWuqianghao, 2020-11-19:
Alright, guess I will just have to stick with Hackintosh for this period of time.

closed 2020-11-19 by KenWuqianghao