simprecicchiani/ThinkPad-T460s-macOS-OpenCore #55: BLUETOOTH: NO INFORMATION ABOUT

taktiker12, 2020-11-15:
THANK U FOR YOUR EFFORT ; I JUST WANT TO SIGNALIZE THAT THE WIFI IS WORKING PERFECTLY BUT NO SIGN FROM BLUETOOTH;I AM USING INTEL AC 8260 ON X260 . 
ANY SUGGESTIONS IS WELCOME
GRAZZI

KenWuqianghao, 2020-11-16:
Did you enable kext in the configlist illustrated in post installation?

mhl221135, 2020-11-16:
I also have this trouble i fix it using last kexts from here https://github.com/OpenIntelWireless/IntelBluetoothFirmware

Also i have my SecureBootModel=Disabled

taktiker12, 2020-11-16:
thank you for your replies; I just solved the issue today .
I had linux installed in the same disck , I had for days disabled bluetooth on it . Today I booted linux and reenable the bluetooth and after that I boot Mac OS and everything worked.

closed 2020-11-17 by simprecicchiani