simprecicchiani/ThinkPad-T460s-macOS-OpenCore #57: Which ssd 512 do you recomend for this laptop ?

mhl221135, 2020-11-20:


simprecicchiani, 2020-11-21:
I recently installed a WD black sn750 and it works just fine. Any other ssd will work just fine but remember to avoid Samsung’s PM981 and PM961.

mhl221135, 2020-11-22:
> I recently installed a WD black sn750 and it works just fine. Any other ssd will work just fine but remember to avoid Samsung’s PM981 and PM961.

Thanks, today buy sn750 and its working fine )

closed 2020-11-22 by mhl221135

hirwa145, 2020-11-25:
Is there noticeable improvement from using previous M.2 Sata SSD you had before?

mhl221135, 2020-11-25:
> Is there noticeable improvement from using previous M.2 Sata SSD you had before?

I timed the start time of big sur and it changed by ~ 20 seconds on sata ssd it was 1:25 and on nvme 0:58

simprecicchiani, 2020-11-25:
> Is there noticeable improvement from using previous M.2 Sata SSD you had before?

No, in daily task not at all. I upgraded only because I had an NVMe lying around after selling my desktop.