simprecicchiani/ThinkPad-T460s-macOS-OpenCore #59: macOS Big Sur 11.0.1 (20B29) - Battery Service Reccomended

nicogig, 2020-11-21:
Hello,

I recently reinstalled macOS on my ThinkPad T460s. Everything is working fine, including the DW1820a and bluetooth strangely enough!

The only issue I am facing is that the macOS Battery Applet says that my battery needs replacing. I checked it on HWMonitor but it seems to be in good conditions still, with 293 cycles out of 1000 and a max capacity of 3105 mAh. I should also note that neither Windows or Linux have previously told me that the battery needs replacing, nor has Lenovo's Vantage utility.

Any ideas why I am getting this weird issue?

mhl221135, 2020-11-23:
You can just ignore it, i also have this message but all working fine (i have 2500~mAh ) on my battaries

cty12, 2020-11-28:
This is because your battery health percentage is below a certain threshold. The design capacity on my machine is `4349 mAh`. The warning will go away if you replace the batteries.

mhl221135, 2021-04-10:
Weared battery (have same notification) it is not the EFI issue

closed 2021-04-10 by mhl221135

nicogig, 2021-04-12:
> Weared battery (have same notification) it is not the EFI issue

It isn't, not in my case. I changed my batteries in Sept 2020. Installing Linux I noticed that the battery information sent to the OS was completely wrong. Recalibrating fixed it. Maybe it isn't an EFI issue but somewhere down the line something is messing up the battery info.

k4kanwar, 2021-12-30:
any solutions to this? @mhl221135

mhl221135, 2021-12-30:
> any solutions to this? @mhl221135

Only batteries change- but you can also live with this message its just talking that batteries max capacity lower than design capacity