simprecicchiani/ThinkPad-T460s-macOS-OpenCore #62: Broadcom 943224PCIEBT2

santokeenan, 2020-11-28:
Hi, thankyou for the instruction and file.
I already installed big sur on my t460s, use intel wifi. All running well, then i replaced my wifi card with brooadcom 943224PCIEBT2, but cannot turn on wifi.
On system information - PCI it already detected but no driver installed. 
What can i do?
Thank you

simprecicchiani, 2020-11-29:
Please check that you have everything in place and your card is working. System information is injected with open core device properties so it’s not a reliable way to check your installation.

santokeenan, 2020-11-29:
thank you for your response, its work with kext AirportBrcmFixup and BrcmPatchRAM3.
But i cannot get airdrop and sidecar features

closed 2020-11-30 by santokeenan