simprecicchiani/ThinkPad-T460s-macOS-OpenCore #63: Hello, i find the way to little speed up boot process for those who use intel wifi cards

mhl221135, 2020-11-28:
Hello, i find the way to little speed up boot process for those who use intel wifi
https://github.com/OpenIntelWireless/itlwm/issues/353

If your card is AC8260 you need to replace 

`find itlwm/firmware/ -type f ! -name 'iwm-7265-*' -delete`

by

`find itlwm/firmware/ -type f ! -name 'iwm-8000C*' -delete`

Here is ready to use kexts for BigSur  
[itlwm and airportitlwm kexts.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5610989/itlwm.and.airportitlwm.kexts.zip)

P.S. Use only one kext in the same time( i recomend to use `Airportitlwm.kext`)

_Originally posted by @mhl221135 in https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/issues/48#issuecomment-735268459_

simprecicchiani, 2020-11-30:
Thanks! You can open a PR and add that on intel wifi section of README.md if you'd like.

closed 2020-12-01 by mhl221135

iamUmerAkbar, 2022-08-07:
hey, can you please add Airportitlwm.kext  for AC8260 monterey??

iamUmerAkbar, 2022-08-07:
> Hello, i find the way to little speed up boot process for those who use intel wifi [OpenIntelWireless/itlwm#353](https://github.com/OpenIntelWireless/itlwm/issues/353)
> 
> If your card is AC8260 you need to replace
> 
> `find itlwm/firmware/ -type f ! -name 'iwm-7265-*' -delete`
> 
> by
> 
> `find itlwm/firmware/ -type f ! -name 'iwm-8000C*' -delete`
> 
> Here is ready to use kexts for BigSur [itlwm and airportitlwm kexts.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5610989/itlwm.and.airportitlwm.kexts.zip)
> 
> P.S. Use only one kext in the same time( i recomend to use `Airportitlwm.kext`)
> 
> _Originally posted by @mhl221135 in [#48 (comment)](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/issues/48#issuecomment-735268459)_

hey my card is AC8260 can you make Airportitlwm.kext for monterey