simprecicchiani/ThinkPad-T460s-macOS-OpenCore #64: Trackpad, Trackpoint and physical buttons

zxcvb5788, 2020-11-29:
hello，bro，first thanks for your EFI，My laptop is Thinkpad x1c 2016, the computer configuration (I7 6600U, 16G, 2K, intel HD520) is similar to your T460S, your EFI file is almost perfect on my computer（catalina 10.15.7）, but there is a problem, Trackpad, Trackpoint and physical buttons after wake up from sleep There will be no response (the keyboard and mouse work normally). Do you have this problem?

zxcvb5788, 2020-11-29:
I booted the system with a USB flash drive. Do you think this problem is caused by the automatic uninstallation of the USB flash drive during sleep? Because in addition to wake-up from sleep, Trackpad, Trackpoint and physical buttons work normally

simprecicchiani, 2020-11-30:
There should be a bug in the trackpad driver. 
try to set `sudo pmset -a standby 0` in terminal.

seanliu96, 2021-03-31:
I still have the same problem with my T460s. I tried the command above but it did not work.

zxcvb5788, 2021-04-01:
> I still have the same problem with my T460s. I tried the command above but it did not work.

I solved it by using “VoodooPS2Controller.kext” 

[image]

seanliu96, 2021-04-01:
> > I still have the same problem with my T460s. I tried the command above but it did not work.
> 
> I solved it by using “VoodooPS2Controller.kext”
> 
> [image]

Hi, thanks for your reply. I also used this kext, but the trackpad still occasionally failed to work after sleep. Haven't you encountered this problem?

zxcvb5788, 2021-04-01:
> Hi, thanks for your reply. I also used this kext, but the trackpad still occasionally failed to work after sleep. Haven't you encountered this problem?

I don't have this problem again after i use “VoodooPS2Controller.kext”, even sleep 2 days.  Is keyboard working correctly when the trackpad didn't work? Maybe you can provide an email and I'll send my VoodooPS2Controller.kext to you

zxcvb5788, 2021-04-01:
> Hi, thanks for your reply. I also used this kext, but the trackpad still occasionally failed to work after sleep. Haven't you encountered this problem?

讲中文？

seanliu96, 2021-04-01:
> I don't have this problem again after i use “VoodooPS2Controller.kext”, even sleep 2 days. Is keyboard working correctly when the trackpad didn't work? Maybe you can provide an email and I'll send [...]

You are so kind. My email is seanliu96@outlook.com . Thanks.

seanliu96, 2021-04-01:
> 讲中文？

可以。但是你的办法可能可以帮助到其他国家的一些朋友。

Okay. But maybe guys from other countries can benefit from your solution.

seanliu96, 2021-04-02:
Seems solved. Thank @zxcvb5788 for providing his kext. Currently, the 2.1.7 version of VoodooPS2Controller.kext works well. If someone has the same problem, please try https://github.com/acidanthera/VoodooPS2/releases/download/2.1.7/VoodooPS2Controller-2.1.7-RELEASE.zip

mhl221135, 2021-04-02:
> Seems solved. Thank @zxcvb5788 for providing his kext. Currently, the 2.1.7 version of VoodooPS2Controller.kext works well. If someone has the same problem, please try https://github.com/acidanthera/VoodooPS2/releases/download/2.1.7/VoodooPS2Controller-2.1.7-RELEASE.zip

you using hiberinate mode ?

seanliu96, 2021-04-04:
> > Seems solved. Thank @zxcvb5788 for providing his kext. Currently, the 2.1.7 version of VoodooPS2Controller.kext works well. If someone has the same problem, please try https://github.com/acidanthera/VoodooPS2/releases/download/2.1.7/VoodooPS2Controller-2.1.7-RELEASE.zip
> 
> you using hiberinate mode ?

I am not sure whether mac os the support hiberinate mode.Previously, I encountered the bug when my laptop automatically slept.

mhl221135, 2021-04-10:
Resolved in latest EFI update sleep-wake working fine

closed 2021-04-10 by mhl221135