simprecicchiani/ThinkPad-T460s-macOS-OpenCore #67: Thinkpad X260 with 2 Batteries

taktiker12, 2020-12-03:
ciao siempri
thank you for your great contribution to help us using your work ; well , I am on Thinkpad x260 i5 6300U with internal and external battery . I am confused about the battery status that is showing only one battery . so I have a question , shall I modify something to get 2 batteries showing on the status bar ?
thank You

simprecicchiani, 2020-12-03:
Try to install HWmonitor and see what it shows.

taktiker12, 2020-12-03:
thank you for reply and advice .I will install hwmonitor

taktiker12, 2020-12-03:
ok, it shows under batteries only internal battery

taktiker12, 2020-12-03:
[image]

mhl221135, 2021-04-10:
You cant get 2 batteries showed in your bar (this efi uses patch that combines 2 batteries as one logical)

closed 2021-04-10 by mhl221135