simprecicchiani/ThinkPad-T460s-macOS-OpenCore #68: cmos checksum bad after long sleep

mhl221135, 2020-12-04:
After long sleep on last efi and last release Big Sur i get "cmos checksum bad"
Then when i press "press esc to continue boot" i see opencore picket and when i choose my os is is showing me black screen without any verbouse

Then i hold pwr button and restart laptop and on this start up cycle everything working fine

What can it be?

mhl221135, 2020-12-11:
[image]

NotNocky, 2020-12-11:
I got something similar but it was after the laptop had died in sleep

simprecicchiani, 2020-12-12:
Did you had standby turned on or off?

findyy99, 2020-12-16:
I am T450S, I have the same question, I have enable `DisableRtcChecksum`.Does't work.

mhl221135, 2020-12-18:
Hello, and thank you for your work, i disable stand by and update my efi to 0.3.4 and i dont see this error anymore

> Did you had standby turned on or off?

closed 2020-12-18 by mhl221135