simprecicchiani/ThinkPad-T460s-macOS-OpenCore #71: [help wanted] Opencore 0.6.4 with YogaSMC

xuandung38, 2020-12-08:
I find YogaSMC to have more superior features than ThinkpadAssistant. For example the Fan control, Charge Threshold Management ... really useful. Would you mind spending some of your precious time trying to research it?

aptonline, 2020-12-09:
I second this, YogaSMC looks great, especially the fan control abilities. @xuandung38 my understanding is that this replaced VirtualSMC is this correct?

xuandung38, 2020-12-09:
No! YogaSMC based on VirtualSMC. So we allway need VirtualSMC and load it before YogaSMC.

xuandung38, 2020-12-09:
I tried YogaSMC, it works very well. But I don't know how to fix sleep so I'm still temporarily using this repo. If you need it I will send it to you.

simprecicchiani, 2020-12-10:
Hi, I did some researches but unfortunately I don't have the knowledge and time to implement this right now.

aptonline, 2020-12-10:
maybe @xuandung38 can point you in the right direction as what needs to be done? from what @xuandung38  is saying its a KEXT and System preferences pane that runs alongside VirtualSMC and can be used instead of ThinkAssistant so in essence its about adding the KEXT Into the EFI built with a supplementary step of adding the preference pane afterwards.

xuandung38, 2020-12-10:
Here is my EFI OC 0.6.3 with yogaSMC. 
I just don't know how to fix sleep for it.

You can take it and researches

[T460SxYogaSMC.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5674774/T460SxYogaSMC.zip)

simprecicchiani, 2020-12-11:
There’s no need for supplementary SSDTs?

xuandung38, 2020-12-11:
No, we need to add some additional SSDT to manage PIN, Fan and LED control.

xuandung38, 2020-12-11:
Guide:

xuandung38, 2020-12-11:
Here is guide:
Change ERAM in method RE1B to ERM2 or any.
Add 2 method to SSDT_BAT0 like this
[image]
Download [Latest release](https://github.com/zhen-zen/YogaSMC/releases/tag/1.3.0)
Compile and add SSD-DYTC-Tx80.dsl + SSD-THINK.dsl to ACPI

Remove all __QXX_ method in ACPI->patch

Add YogaSMC.kext to kext and kernel -> done

simprecicchiani, 2020-12-12:
Have you already done that? Does it fix sleep issues?

xuandung38, 2020-12-13:
When i use repatch your ACPI it work well. But led control not work , i'm researching.

[image]
[EFI.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5684394/EFI.zip)

xuandung38, 2020-12-13:
try it and tell me.

simprecicchiani, 2020-12-18:
I made some changes to the original ACPI patches from OC-Little, then I looked into @tylernguyen x1c6 config to see how he implemented the tool.
I'm still not confident with ACPI but it seems that most of YogaSMC's functions are working.
Some LEDs are fully functional, others are partially supported (i.e. charging light can't pulse, only blink).
[EFI.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5718398/EFI.zip)

xuandung38, 2020-12-19:
> I made some changes to the original ACPI patches from OC-Little, then I looked into @tylernguyen x1c6 config to see how he implemented the tool.
> I'm still not confident with ACPI but it seems that most of YogaSMC's functions are working.
> Some LEDs are fully functional, others are partially supported (i.e. charging light can't pulse, only blink).
> [EFI.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/5718406/EFI.zip)
> 
> Edit: I also added BrightnessKeys.kext

Let's me test it!

xuandung38, 2020-12-21:
Everythings is ok , but Charge Threshold function notwork.

aptonline, 2020-12-23:
Progress on this is looking good, noticed new version available with a ton of improvements: https://github.com/zhen-zen/YogaSMC/releases/tag/1.4.0

simprecicchiani, 2020-12-23:
> Progress on this is looking good, noticed new version available with a ton of improvements: https://github.com/zhen-zen/YogaSMC/releases/tag/1.4.0

Led control and Mic button broke with this release

xuandung38, 2020-12-25:
Hi @simprecicchiani ! I have been using your efi so far I have not seen any inconvenience or bugs, I think it's time to release a separate branch for this version.

aptonline, 2020-12-25:
> Led control and Mic button broke with this release

Does this version fix it?

https://github.com/zhen-zen/YogaSMC/releases/tag/1.4.1

simprecicchiani, 2020-12-26:
> Does this version fix it?
> 
> https://github.com/zhen-zen/YogaSMC/releases/tag/1.4.1

Yes it does, everything works as expected.

> Hi @simprecicchiani ! I have been using your efi so far I have not seen any inconvenience or bugs, I think it's time to release a separate branch for this version.

I was thinking of releasing it with the next OpenCore update. I was quite happy with ThinkpadAssistant but, from what I understood, it will be soon deprecated. I don't think there is a need for two separate configs.

alexdelli, 2020-12-29:
Why I can't see the YogaSMC icon?

xuandung38, 2021-01-05:
OpenCore 0.6.5 released. let's do it <3

simprecicchiani, 2021-01-05:
> OpenCore 0.6.5 released. let's do it <3

#82

mhl221135, 2021-01-08:
This thing not working its jumping to balanced and not changing, and what it is performance modes ?)