simprecicchiani/ThinkPad-T460s-macOS-OpenCore #72: some kexts dont load upon waking from sleep

NotNocky, 2020-12-09:
using an alpha version of the intel airport kexts, but the ones you include also do the same thing as well as the track pad 

its not really consistent and i cant always reproduce it but i know its from putting the machine to sleep

let me know if theres any logs etc you need

simprecicchiani, 2020-12-09:
Unfortunately yes, this is a common issue. I should update the README. A quick fix is to completely disable standby by `sudo pmset -a standby 0`. Doing so will prevent the laptop to go into hibernation and avoid any related trouble. If you are wondering, yes, battery will decrease a bit more during sleep, but its acceptable imho.

NotNocky, 2020-12-11:
Yeah thats a much better alternative, how would you set that perminatly? doing it after each reboot is a pain

simprecicchiani, 2020-12-12:
Why? Does it revert to 1 every time? That’s very strange, could you please post your pm settings with ‘pmset -g’?

vale981, 2020-12-19:
I've updated the touchpad kexts and waking up from sleep seems to work.

duszmox, 2021-04-09:
This issue should be solved in release 0.6.8, thanks to @mhl221135

mhl221135, 2021-04-10:
Now with latest EFI and latest airportitlwm.kext no issues after wake from sleep did you try to update?

> using an alpha version of the intel airport kexts, but the ones you include also do the same thing as well as the track pad
> 
> its not really consistent and i cant always reproduce it but i know its from putting the machine to sleep
> 
> let me know if theres any logs etc you need

> Yeah thats a much better alternative, how would you set that perminatly? doing it after each reboot is a pain

mhl221135, 2021-04-21:
Fixed

closed 2021-04-21 by mhl221135