simprecicchiani/ThinkPad-T460s-macOS-OpenCore #74: Two questions before beginning

MrksHfmn, 2020-12-12:
Thanks a lot first of all for this repository! I will try to install Mac OS on my T460s in the next few days. I have already created a USB flash drive with the help of this tool: [Hackintosh-linux-Installer](https://github.com/Broly1/Hackintosh-linux-Installer)
It automatically creates an EFI partion with EFI/Boot and EFI/OC. Can I now take your release and copy it into the EFI folder, or should the folder be completely replaced with your release?
Would you install Bug Sur or Catalina initial? Thanks a lot!

simprecicchiani, 2020-12-12:
I would replace the entire EFI folder to be sure everything is in place. 
If you’re planning to update to Big Sur then it’s better to install it straight away.
To be honest if you don’t need any new functionality I think Catalina performs better at the moment. I kinda regret having updated to 11.0.1 on release date.
Last note. I think it’s better, not mandatory, to generate SMBIOS infos and copy them into config.plist before starting the installation.

taktiker12, 2020-12-15:
`i agree with simpreccianni that catalina is the best at the moment ..mojave is also good`

MrksHfmn, 2020-12-15:
I have managed to install Big Sur on the T460s. As far as I can tell, everything runs quite well. Just need another wifi card. How is Catalina better than Big Sur?

taktiker12, 2020-12-16:
On Big Sur Microsoft Word take more time for loading than catalina and
mojave.. From login to desktop too... I think big sur is an experimental
Mac OS...

simprecicchiani, 2020-12-17:
Microsoft Office is quite bad on macOS regardless of which version you’re running. Poor performance and lack of features. If you’re a heavy Office user you better running it on Windows.

taktiker12, 2020-12-17:
i am unix user since the begin of 90's ..microsoft word is needed in rarely special cases.

closed 2020-12-19 by simprecicchiani