simprecicchiani/ThinkPad-T460s-macOS-OpenCore #77: Unknown Startup Message

KenWuqianghao, 2020-12-14:
I received this message saying "OCS: Failed to calculate size of false field containing <empty> as type integer, context <PlayChime>!". The startup was perfectly fine and working. I'm just wondering if this is some potential issue I need to fix or recommended to solve. 
[image]

cdpidan, 2020-12-15:
Change UEFI-> Audio-> PlayChime:Auto

KenWuqianghao, 2020-12-15:
Solved, thank you very much

closed 2020-12-15 by KenWuqianghao

simprecicchiani, 2020-12-15:
> Solved, thank you very much

Was it set differently in release's config.plist?

KenWuqianghao, 2020-12-15:
> > Solved, thank you very much
> 
> Was it set differently in release's config.plist?

Yes it was

simprecicchiani, 2020-12-15:
Ok thanks, I'll fix it in the next release.