simprecicchiani/ThinkPad-T460s-macOS-OpenCore #78: Hackintosh green screen, maybe because the RAM card is not fixed in the slot

KenWuqianghao, 2020-12-17:
My hardware is 512GB SSD, 4GB soldered plus 16GB RAM, i7 6th gen. The system won't continue working or startup if I don't use the solution below.

I've found a bizarre solution through this issue which is holding the extra RAM card tightly to the RAM slot and the system would start up and function normally, so I put a stack My hardware is 512GB SSD, 4GB soldered plus 16GB RAM, i7 6th gen. The system won't continue working or startup whenever there are small collisions or turbulence. Does anyone have a similar issue or has a proper solution for it?

mhl221135, 2020-12-18:
it looks like the contact of the RAM disappears with jolts, you can try to clean its contacts with a pencil eraser and insert it correctly until it clicks

KenWuqianghao, 2020-12-18:
I had some videos demonstrating my situation even after cleaning the slot. Here they are:
Without Extra RAM: 
[https://e.pcloud.link/publink/show?code=XZGd97Z6LHxuCBP5PRVKBvjxAsIO8pVYgUX(https://e.pcloud.link/publink/show?code=XZGd97Z6LHxuCBP5PRVKBvjxAsIO8pVYgUX)
With RAM:
[https://e.pcloud.link/publink/show?code=XZbq97ZMsKrR3BdCbLMBuzL77glIu8DEMsV](https://e.pcloud.link/publink/show?code=XZbq97ZMsKrR3BdCbLMBuzL77glIu8DEMsV)
With RAM and holding it against the motherboard: [https://e.pcloud.link/publink/showcode=XZRq97Zr2RkkkNjLkuVM0nupj1Sm4HrpQdk(https://e.pcloud.link/publink/showcode=XZRq97Zr2RkkkNjLkuVM0nupj1Sm4HrpQdk)

simprecicchiani, 2020-12-18:
Have you tried booting another OS? You could try a live Linux distro.

KenWuqianghao, 2020-12-18:
I'm not sure how that will help solve the issue, but I will be trying it now.

KenWuqianghao, 2020-12-19:
Okay, I just extra tried booting from a live Linux distro. Results are the same. Without the RAM or with me pressing the RAM, the OS starts perfectly. If I'm not pressing against the RAM, the issue still occurs, the OS just crashes and display some green lines. At this point, I think it is my hardware issue.
[image]

simprecicchiani, 2020-12-19:
Sorry to hear that, it's very unfortunate. You could give it a try with another ram stick but the slot is likely to be broken.

closed 2020-12-19 by simprecicchiani