simprecicchiani/ThinkPad-T460s-macOS-OpenCore #81: loading bar / verbose output after login

nshooter35, 2020-12-26:
Hi,

Thanks for the great work @simprecicchiani 

I am just having one issue - when I login / enter my password into OSX, a loading bar is diplayed and a load of verbose output displayed. This seems to delay in desktop loading time slightly. I have not seen this behaviour on other hackintosh's I have so I wonder if there is an issue. I have tried a clean install. Using big sur, but was also happening on previous version. 

Any thoughts appreciated. 

thanks

simprecicchiani, 2020-12-26:
Do you have your disk encrypted?

nshooter35, 2020-12-26:
> Do you have your disk encrypted?

I do. Is that expected behaviour with encryption?

nshooter35, 2020-12-26:
Also I'll send over a donation as I think your work is great. The only slight issue is I find the trackpad slightly too sensitive if I have tap to click enabled. Can result in some clicks in error. Not sure if my specific hardware or limitation of kext but I get around it by using a mouse.

simprecicchiani, 2020-12-26:
> I do. Is that expected behaviour with encryption?

Yes, it is. Not the verbose output tho. If you have it disabled it shouldn't output anything after the login screen (a parte from the loading bar).

> The only slight issue is I find the trackpad slightly too sensitive if I have tap to click enabled.

I can confirm that too, I have tap to click disabled because of it. I've never asked VoodooRMI's maintainers about it to be honest. I'll do that sooner or later.

nshooter35, 2020-12-27:
Super thanks for the reply. I'll try turning off encryption to see if it resolves. 

re trackpad - looks like similar issue posted here:
https://github.com/VoodooSMBus/VoodooRMI/issues/89