simprecicchiani/ThinkPad-T460s-macOS-OpenCore #89: yogaSMC Charge threshold & DYTC function switcher not working (latest 6.5 EFI)

mhl221135, 2021-01-19:
1) Charge threshold not working at al, i tried to set different statments but if stop value is 70 for example it is still charging battery
2) DYTC function switcher not svitching from balanced position

With charge threshold i tried this, and this not change nothing same as the yoga panel
``
ioio -s ThinkVPC Battery 0
ioio -s ThinkVPC setCMstart 60
ioio -s ThinkVPC setCMstop 70
ioio -s ThinkVPC Battery 1
ioio -s ThinkVPC setCMstart 60
ioio -s ThinkVPC setCMstop 70
``

mhl221135, 2021-01-20:
I checked and in BATX implementation the control of the charging threshold works (only the percentage of the SMC sensors from smchwmonitor2.app and the MacOS panel  slightly differs and sometimes it stop charging a little after stop threshold for example my stop threshold is 80% and laptop exacly stop charging at 85%), 
maybe it would be better to implement BATX it instead of your main method for the battery in your repo?

DYTC switcher still not working, maybe there is some solution for it on benbenders repo?

simprecicchiani, 2021-01-22:
So BATX work fo you then?

mhl221135, 2021-01-22:
> So BATX works for you then?
> 
> Concerning DYTC, as other reported there is a performance loss with respect to Intel thermal solution (About 100 CB23 points in multi-core). I don't know how to implement control from YogaSMC though. 

Yes BATX working perfect with threshold

closed 2021-04-02 by mhl221135