simprecicchiani/ThinkPad-T460s-macOS-OpenCore #93: The fans are noticably louder then on Windows/Linux

wrobeljakub, 2021-02-04:
The fans are running more often then on other operating systems and they are quite load. I'm not sure is this a problem with my machine or not. Could someone verify that?

rsdev69, 2021-02-04:
> The fans are running more often then on other operating systems and they are quite load. I'm not sure is this a problem with my machine or not. Could someone verify that?

This is normal, this processor sucks Big Sur)

wrobeljakub, 2021-02-04:
> > The fans are running more often then on other operating systems and they are quite load. I'm not sure is this a problem with my machine or not. Could someone verify that?
> 
> This is normal, this processor sucks Big Sur)

Is there a way to change that? For example lower the clock speeds or change the fan curve using YogaSMC.

rsdev69, 2021-02-04:
> Is there a way to change that? For example lower the clock speeds or change the fan curve using YogaSMC.

Cpu is terrible. This is my mind.

alexdelli, 2021-02-05:
I've the I7 and after change the thermal paste, fans are very quiet (ah and don't forget to calibrate the cpu usage with CPUFriend)

MarkMorrow, 2021-02-05:
I have T460s i5, and part of what I like about it is no fan noise and little heat wether running Win10 or Big Sur.  This is compared to my other laptops, 15" MacBook Pro i7 (still on Mojave) and work's T480 i7.  I wish the T460s had the display of the MacBook, but cost was right, and perfect for many situations.  I suspect the i5 is the main difference for this.

I second @alexdelli 's thoughts that your thermal paste may need refreshing or the fan cleaning, especially if yours is i5.

wrobeljakub, 2021-02-05:
> I've the I7 and after change the thermal paste, fans are very quiet (ah and don't forget to calibrate the cpu usage with CPUFriend)

I've tried to create more power saving plan in CPUFriend but without any luck. I'll try to do the repaste and I'll be back with the results.

wrobeljakub, 2021-02-05:
> I have T460s i5, and part of what I like about it is no fan noise and little heat whether running Win10 or Big Sur.  This is compared to my other laptops, 15" MacBook Pro i7 (still on Mojave) and work's T480 i7 (Win10).  I wish the T460s had the display of the MacBook, but cost was right, and perfect for many situations.  I suspect the i5 is the main difference for this.
> 
> I second @alexdelli 's thoughts that your thermal paste may need refreshing (or perhaps the fan needs a good cleaning), especially if yours is i5.

I have the i5-6300u. I was even thinking of applying liquid metal to improve the thermals.

simprecicchiani, 2021-02-07:
I found mine (i7) reasonably silent when room temp is <20°C

> I have the i5-6300u. I was even thinking of applying liquid metal to improve the thermals.

Repasting a laptop is ~~always~~ a good idea. I've thought about liquid metal but from what I understood it needs to be redone every year or so.

closed 2021-02-12 by wrobeljakub