simprecicchiani/ThinkPad-T460s-macOS-OpenCore #94: Dual system win10 recognized as wrong modle and serial number

559911, 2021-02-05:
I’m using dual system on my t460s with one ssd，MacOS works great but under the windows10 my computer was recognized as ‘MacBook pro’ and the software Lenovo vantage can’t update my driver(because of the wrong model type and serial number ) is there any way to fix it?
I find that if I boot from the boot menu(press f12 and choose my ssd)then win10 has a right model .

MarkMorrow, 2021-02-05:
I also have dual boot (Big Sur and Win10) on single internal SSD within my T460s.  My goal was to use Parallels to access the Win10 "BootCamp" partition from Big Sur, and also boot directly from OpenCore's boot menu.  Slightly different than your request, but perhaps this will still help.  I found the OpenCore section of this webpage helpful in setting it up:
[https://www.reddit.com/r/hackintosh/comments/gs0s1o/create_duelboot_windows_on_hackintosh_drive/](https://www.reddit.com/r/hackintosh/comments/gs0s1o/create_duelboot_windows_on_hackintosh_drive/)

I started with a system running High Sierra and Win10 as a true BootCamp install, and upgraded it to Big Sur.  After that Win10 would not show from the OpenCore's boot menu but I could still be accessed from Parallels.  Unfortunately activation was broken, so ultimately I ended up backing up the EFI , and then  deleting Win10 and reinstalling it using a Win10 USB (i.e. EFI; not BootCamp install).  That worked and activate using OEM Win10 activation.  I  then adding OpenCore back to the EFI, which allowed Parallels to access Win10 (activated) :), BUT Win10 was still not showing within OpenCore's boot menu, so like you, I had to use F12.  I wanted it in the OpenCore menu, so that is where the link I provided comes in.  The OpenCore section toward the end helped me add the entry for Win10 into OpenCore's config.plist, so now Win10 shows and boots. Lenovo has no issue recognizing it to install drivers.  The moral of the story, is that how Win10 was installed likely makes a difference.  Hopefully there is some crumb of help within this.  Good Luck.

559911, 2021-02-05:
Thank you for your detailed answers！
I'll have a try.

MarkMorrow, 2021-02-05:
I may have mislead you.  I was using Lenovo Service Bridge, and then running Lenovo System Update, which loaded all specific drivers but does not specifically state system details.  Just now I installed Lenovo Vantage from the Windows Store and see what you mean.  When booted from OpenCore it shows it as a "MacBook Pro" with a product number "MacBookPro131", Bios version: 429.60.3.0.0, and the serial number set by OpenCore.  Perhaps now that I'm booting from OpenCore I'm not getting any further updates.

It seems that OpenCore needs selective ACPI patching/SMBIOS when Win10 is picked.  I'm unsure how to do that, and fear that tampering with that could de-activate Win10.  Although reactivation should be possible, Parallel's might then have issues with the change and not be able to reactivate as OEM.  I thing I will leave it as is and apply updates manually.

That said the little research I did found a discussion with this statement:

> _The "correct" way to do ACPI patching without breaking other operating systems is to use **If _OSI("Darwin") {}** in your SSDTs to selectively enable patches for macOS only_

Perhaps someone with more experience can chime in.

559911, 2021-02-05:
> I may have mislead you. I was using Lenovo Service Bridge, and then running Lenovo System Update, which loaded all specific drivers but does not specifically state system details. Just now I [...]

Thanks, hope OpenCore updated  can fix it! I'll try Lenovo Service Sridge

simprecicchiani, 2021-02-07:
> > _The "correct" way to do ACPI patching without breaking other operating systems is to use **If _OSI("Darwin") {}** in your SSDTs to selectively enable patches for macOS only_
> 
> Perhaps someone with more experience can chime in.

Our SSDTs should already be "OS sensitive". 
To prevent SMBIOS infos to get passed to Windows set:
- `Kernel -> Quirks -> CustomSMBIOSGuid` : `True`
- `PlatformInfo -> UpdateSMBIOSMode` : `Custom`
Unfortunately, this doesn't make Lenovo Vantage to work.

559911, 2021-02-07:
> Our SSDTs should already be "OS sensitive".
> To prevent SMBIOS infos to get passed to Windows set:
>
> * `Kernel -> Quirks -> CustomSMBIOSGuid` : `True`
> * `PlatformInfo -> UpdateSMBIOSMode` : `Custom` [...]

Now I get the right serial number.

559911, 2021-02-08:
>w > > _The "correct" way to do ACPI patching without breaking other operating systems is to use **If _OSI("Darwin") {}** in your SSDTs to selectively enable patches for macOS only_
> > 
> > 
> > Perhaps someone with more experience can chime in.
> 
> Our SSDTs should already be "OS sensitive".
> To prevent SMBIOS infos to get passed to Windows set:
> 
> * `Kernel -> Quirks -> CustomSMBIOSGuid` : `True`
> * `PlatformInfo -> UpdateSMBIOSMode` : `Custom`
>   Unfortunately, this doesn't make Lenovo Vantage to work.
I reinstall the Lenovo vantage and lenovo active it work！thank you！
[image]

> Our SSDTs should already be "OS sensitive".
> To prevent SMBIOS infos to get passed to Windows set:
>
> * `Kernel -> Quirks -> CustomSMBIOSGuid` : `True`
> * `PlatformInfo -> UpdateSMBIOSMode` : `Custom` [...]

 I reinstall the Lenovo vantage and lenovo active protection system ,the lenovo vantage works!
thank you !
[image]

closed 2021-02-08 by 559911

MarkMorrow, 2021-02-08:
> Our SSDTs should already be "OS sensitive".
> To prevent SMBIOS infos to get passed to Windows set:
> 
> * `Kernel -> Quirks -> CustomSMBIOSGuid` : `True`
> * `PlatformInfo -> UpdateSMBIOSMode` : `Custom`

Confirmed.  That worked for me; not sure how it got set to false, but after enabling the Quirk Wn10 recognizes it as T460s and finds Lenovo updates.  Also remained activated when booted from OC menu.

Parallels from Big Sur is another story.  It shows not activated, but it appears it wasn't before either (just thought it was for it was within 60 day grace).  May not be possible to activate it again with OEM key, but likely out of scope of this thread.

Thanks you!!!