simprecicchiani/ThinkPad-T460s-macOS-OpenCore #95: 0.6.6 release Empty Drivers folder & not all files in the resource folder

mhl221135, 2021-02-08:
There is empty Drivers folder in latest release zip so system halting on not existing HFS driver

Also picket is in the text mode fixed by replacing resources by this [https://github.com/acidanthera/OcBinaryData/tree/master/Resources](url)

after adding drivers and replacing resources all working flowlesly

note: i find info about resources and difference in OC 6.5|6.6 here [insanelymac.com/forum/topic/346592-opencore-065-066-differences/](url)

simprecicchiani, 2021-02-08:
My bad, I must have moved the driver instead of copying them.

closed 2021-02-08 by simprecicchiani

mhl221135, 2021-02-08:
> It should be fixed now, let me know if you guys have any issue.

Using EFI whole day no issues detected, thank you for your great job!