simprecicchiani/ThinkPad-T460s-macOS-OpenCore #96: Bluetooth Kext's gone from Config.plist --> Bluetooth not running

chris8205, 2021-02-10:
Hi, 

in your preconfigured Config.plist for the T460s the Bluetooth support is not enabled.
--> BT Icon showed but not running / discovering BT devices..

i've addet those two Kexts manually and all is running fine..

was there an issue why you've removed it?

best whishes

mhl221135, 2021-02-10:
> Hi, 
> 
> in your preconfigured Config.plist for the T460s the Bluetooth support is not enabled.
> --> BT Icon showed but not running / discovering BT devices..
> 
> i've addet those two Kexts manually and all is running fine..
> 
> was there an issue why you've removed it?
> 
> best whishes 

Ita removed becouse most users using native broadcom cards

But yoy can find manual under readme>postinstall>intel wlan cards

simprecicchiani, 2021-02-11:
Correct

closed 2021-02-11 by simprecicchiani