simprecicchiani/ThinkPad-T460s-macOS-OpenCore #99: OpenCorePkg  release 0.6.7

iTer99, 2021-03-05:
update 0.6.7 please and thank you so much for continue this project

simprecicchiani, 2021-03-05:
[There you go](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/releases/tag/0.6.7).
Haven't tested it though.

iTer99, 2021-03-06:
> [Của bạn đây](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/releases/tag/0.6.7) .
> Chưa thử nghiệm nó mặc dù.
> 
> Chỉnh sửa: Đã kiểm tra, đang hoạt động.

thank you so much, this working very good

closed 2021-03-06 by simprecicchiani

MarkMorrow, 2021-03-06:
The serial number that shows in the About screen is now different.  I transferred over the entries in /EFI/OC/Config.plist within PlatformInfo -> Generic and double checked it is correct.  Messaging, iCloud and Apple watch unlock still working, so not sure if this is a feature or and issue.  Thoughts?

simprecicchiani, 2021-03-06:
I didn't noticed that. Mine is showing a different SN too. I'll take a look at OpenCore's changelog closely.

MarkMorrow, 2021-03-06:
Thanks much. Glad it is not just me.  I just opened a new issue (since this one is closed).  Came back to remove my post here, but will leave it since you have already replied.

MarkMorrow, 2021-04-09:
Fixed in v0.6.8 in @duszmox 's fork:
[https://github.com/duszmox/Thinkpad-T460s-macOS-OpenCore](https://github.com/duszmox/Thinkpad-T460s-macOS-OpenCore)